SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #2: BIOS Required readme file incomplete

BooBerry, 2020-11-23:
There's more BIOS files required to get things like Intellivision working. Having a full list of all those BIOS files + having the hashes for each file (to verify each file) would be great to have.

The README.md does touch upon the two BIOS files needed to get Intellivision working, however it doesn't tell the user what they're supposed to be named. Adding the BIOS information to that file too would be great.

tjanas, 2020-11-23:
@BooBerry you can get some information for most of these BIOS files from this zip file: http://blog.kevtris.org/blogfiles/ntm_firmware_verJB2.0.zip

[This pull request](https://github.com/SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak/pull/3) added checksum info for almost all of them (except Intellivision and SMS). There are so many SMS BIOS variants, you can read about them at [nointro](https://datomatic.no-intro.org/index.php?page=search&op=datset&s=26&sel_s=Sega+-+Master+System+-+Mark+III&stext=%5BBIOS%5D&where=1&searchme=Search&sort=Title&order=Ascending)

BooBerry, 2020-11-24:
@tjanas Thanks for this! I also was able to figure out the Intellivision stuff and got it working too.

The Intellivision GROM needs to be named `grom.bin` and its CRC32 for the working one I have is `683A4158`.

The Intellivision Executive ROM needs to be named `intvexec1.bin` and its CRC32 for the working one I have is `A85FC6DD`

These need to be added to the readme under the Intellivision section, because it's vague; it doesn't mention what the Executive BIOS needs to be named and its CRC32, same with the GROM binary. The info here should be able to fix that. 👍

frederic-mahe, 2020-11-25:
done in commit d8ae0667f4bbdd4ecfcb9d5fe1a5e0738ce856b6 Thanks @BooBerry and @tjanas

closed 2020-11-25 by frederic-mahe