SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #4: VIce: Project Doom - ROM will not load if 2nd controller is plugged in

Dochartaigh, 2020-11-24:
I have two 8BitDo N30 2.4g retro receivers plugged into my NT Mini Noir (in Player 1 and 2). Only Player 1 controller is actually on (player 2 is off). The game fails to load properly and loads to a screen which says "Stage Select" (although it seems to be an audio test of some sort). 

If I unplug Player 2's retro receiver and re-load the game from the Noir's menu it will load normally. If I plug back in the Player 2 Retro Receiver the Noir locks up and needs to be power cycled to get back up and running (controllers are supposed to be hotswapable while it's on).

Tried with today's jailbreak firmware with the ROM's on a SD card, and through a Krikzz N8 Pro Everdrive cartridge as well.

The ROM is from a SmokeMonster set which I believe is using No-Intro files. Filename is "Vice - Project Doom (USA).nes". Don't have clrmamepro or similar setup currently to get the databases MD5 to match it up, but when I run it from my file I get MD5 checksum of 70edfbe6dc959737199cfb803703a734. This matches a couple hits for this game and the No-Intro set I found online. About 30 other games have played perfect so far. I took a video of this strangeness if you happen to need it. Can upload the exact ROM too if that's allowed.

tjanas, 2020-11-27:
I just tested this game on the original Nt Mini with two original wired NES controllers plugged in and it worked fine.

Dochartaigh, 2020-11-27:
> I just tested this game on the original Nt Mini with two original wired NES controllers plugged in and it worked fine.

Must be something with the new Noir version with it's new FPGA then. I also just tried with 2x wired original controllers (plugged in before even turning the Noir on) and it gives me the same issue. Also tried with a 2.4g plugged into player 1, and a real wired controller in player 2 and same issue.

Spazilton, 2020-11-30:
Contacted OP and verified MD5 on file being used for Vice Project Doom. Cannot replicate on NT Mini V2 with the JB firmware. Works fine on mine. Also works fine loaded through N8 Pro. Tried a variety of controllers plugged into 2nd player port including 2.4 and BT Retro Receivers and regular NES controller.

Dochartaigh, 2020-11-30:
If this helps, another tested this and it works on BOTH his jailbroken Noir, AND his N8 Pro. His MD5 hash is identical to mine. I'm now thinking my Noir has a hardware issue. Anything I can look into besides that? Here's a video I made when I first experienced this issue:

https://youtu.be/mCfmhgU2LXI

I'm going to reinstall the original Noir 1.0 firmware and try it again, then reinstall the jailbreak firmware too and do the same.

thedrdodo, 2020-12-01:
Same behavior happens on the Japanese version of this game named Gun-Dec, on an original cart. On jailbreak firmware 6.2 with two controllers plugged in.

Dochartaigh, 2020-12-04:
Info updated on #19 
(issue with controller compatibility processing/passthrough mode)

tjanas, 2020-12-15:
Please retest with latest firmware v6.5

Elrinth, 2020-12-15:
When using the rom, it is now working correctly for me. I don't own the original so can't test.

tjanas, 2020-12-16:
> Same behavior happens on the Japanese version of this game named Gun-Dec, on an original cart. On jailbreak firmware 6.2 with two controllers plugged into mini v2 noir.

Can you try your cart with the v6.5 firmware?

thedrdodo, 2020-12-18:
Can confirm gun-dec cart works correctly on 6.5 now @tjanas

frederic-mahe, 2020-12-19:
@Dochartaigh can you please confirm the issue is fixed for you too?

Dochartaigh, 2020-12-20:
> @Dochartaigh can you please confirm the issue is fixed for you too?

Just had a minute to load the new firmware - confirmed working on my end. Thank you!

closed 2020-12-20 by frederic-mahe