SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #5: Core Loading Error!

sincethenes, 2020-11-24:
I’m not sure where else to turn. I’ve been looking for a solution and have come up dry. While installing the jailbreak for the NT Mini Noir, we had a three second long brownout in out neighborhood. It was so short, but the Noir didn’t finish the install. I started it again, and it seemingly worked.

The NES core works, but it’s buggy. Invincibility is seemingly turned on randomly in the few games I tested, (Gunsmoke, TMNT, and Panic Restaurant). Also, no other cores are working, giving me the title of this post as an error message (Core Loading Error!)

I tried loading the original BIOS 1.0 on an empty, FAT32 card into the noir, but no dice. The system ignores it and boots like normal into the jailbreak. 

Any help would be appreciated. If this isn’t the place for help, a push in the right direction would be greatly appreciated. Thanks so much!

SmokeMonsterPacks, 2020-11-25:
Did you try re-flashing the Jailbreak?

Also, could you please link to the 1.0 firmware you attempted to flash?

sincethenes, 2020-11-25:
Sure thing. Here is the link to the firmware first off:

https://support.analogue.co/hc/en-us/articles/360052916371-Nt-mini-Noir-Firmware-Update-v1-0

sincethenes, 2020-11-25:
And yes, I did try to reflash.

tjanas, 2020-11-25:
Others have also reported being unable to rollback to original stock firmware from the JB firmware.

sincethenes, 2020-11-25:
Makes me glad to hear I’m not alone in this. Still sucks, but the company is somewhat of a relief.

sincethenes, 2020-11-25:
In the linked issue above, user endrozenberg
had some steps to try. Even though I was sure the console was powered off, unplugging it and plugging it back in seemed to allow it to read the rollback bios just fine. 

Everything looks to be like it was out of the box. I’ll wait a bit to try the jailbreak again just in case. Thanks for your help other user!

closed 2020-11-25 by sincethenes

mistertimely, 2020-11-28:
I am also having the same error screen. However, I did not experience a brown out during flashing. The jailbreak flash completed as expected - with the console resetting itself to the new menu. I powered the console off and swapped to my prepared SD card with the correct bios files and romsets. 

The NES core is working. But when attempting to load other cores, such as Gameboy, Game Gear, Atari 2600, 7800, and Genesis, all display the same "Core Loading Error."

mistertimely, 2020-11-28:
I was able to confirm my issue was due to some unknown problem with my SD card. After using a different SD card, I was able to load other cores.

ZReport, 2020-12-26:
I actually had this issue as well and thought something was up with the SD card I had.  I then realized that I neglected to pull in the "SYSTEM" folder onto the card from the JB zip.  Everyone reading this should first make sure they have pulled in the SYSTEM folder from the included .zip onto their SD card, then check back in case they have a similar issue.