SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #10: Unable to rollback to stock firmware

tjanas, 2020-11-25:
Once Noir JB firmware v6.2 is installed, the Noir [will not accept](https://twitter.com/ecuprofessor/status/1331334458285436928?s=21) rolling back to stock firmware v1.0.

edrozenberg, 2020-11-25:
I was able to roll back successfully some minutes back using https://support.analogue.co/hc/en-us/articles/360052916371-Nt-mini-Noir-Firmware-Update-v1-0 . I'll stay stock for now.

Maybe worth trying to flash again after double-checking a few things -

1. Re-copy the stock firmware to the root of the SD card. Make sure to unmount/eject the SD card properly when done.
2. Try reformatting the SD card (FAT32/vfat) as long as it doesn't contain anything useful, and copying the firmware to it.
3. Try using a different SD card and copying the firmware to it.
4. Try unplugging and replugging the console power adapter, to make sure console is really off.
5. Make sure to wait long enough for the flashing - a few minutes until the LED turns white and the menu comes back again on its own (assuming the reflashing works).

tjanas, 2020-11-25:
Thank you for reporting, I will close this issue.

closed 2020-11-25 by tjanas

edrozenberg, 2020-11-25:
Worked for me but I can't vouch for whether others are all equally lucky :).

JeffHochberg, 2020-12-10:
Hello - I ran into a similar issue but was able to resolve it. I thought I would document it for reference purposes.

_NOTE: These are the steps that worked for me. Your mileage may vary._

From a computer running Windows 10:

1. Insert SD card

2. Open a Command Prompt as Administrator

3. Run the diskpart utility (very helpful utility included with Windows 10)

```
C:\WINDOWS\system32>diskpart

Microsoft DiskPart version 10.0.18362.1171
```

4. List all available disks

```
DISKPART> list disk

  Disk ###  Status         Size     Free     Dyn  Gpt
  --------  -------------  -------  -------  ---  ---
  Disk 0    Online          465 GB   586 MB   *    *
  Disk 1    Online          111 GB      0 B   *    *
  Disk 2    Online         3823 MB  4544 KB
```

5. Select the numeric value of the disk representing your SD Card

_WARNING - MAKE SURE YOU SELECT THE CORRECT DISK - DATA LOSS WILL OCCUR DURING STEP 6!!! DON'T WIPE THE PARTITION ON YOUR HARD DRIVE/SSD! You have been warned!_

`DISKPART> select disk 2`

6. Run the clean command to wipe the partition table from the SD card

`DISKPART> clean`

7. Wait for the message:

`DiskPart succeeded in cleaning the disk.`

8. Quit diskpart and close the Command Prompt

```
DISKPART> exit

Leaving DiskPart...

C:\WINDOWS\system32>exit
```

9. Right-click This Computer on your desktop and choose Manage (or run Computer Management -> Disk Manager)

10. Navigate to your SD Card within Disk Management

11. Right-click the SD Card and choose **New Simple Volume** & create a small partition - 1 GB (1024 MB) is plenty

12. Follow the wizard and format the new partition as FAT (FAT32 should work OK but I selected FAT - aka FAT16)

13. Close Computer Management/Disk Management

14. Copy the official Analogue Nt Mini Noir firmware BIN file to the SD Card then Eject the SD card

15. Power on your Nt Mini Noir - you should notice the LED on the front of the unit flashing - BE PATIENT!

I hope that's helpful!!!