SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #11: Save file not dumping with CopyNES Mini

whitezombie2000, 2020-11-25:
I tried the Dump tool with Dragon Warrior 1 as a test game. I selected Generic MMC1 SxROM.BIN as the mapper. Tick the battery box, press start, the rom dumps without any issue, but the save file is always 0 kb in size when it should be 8 kb. According to the NES Mapper Reader tool that is the correct mapper. The dumped rom can be loaded, played with the NES core without any issue, and creates a proper 8 kb save file.

tjanas, 2020-11-25:
I have a Dragon Warrior cart that I dumped with the old Nt Mini, and that worked fine. I don’t have my Noir hooked up any more, but if no one else confirms this bug I can try to hook it up again and dump the cart.

In the game, after you save, did you try powering off the system and making sure the actual cart itself still sees your game save?

whitezombie2000, 2020-11-25:
Yes the actual cartridge can see and load the save with no issues.

AsherAvery1, 2020-11-27:
I am having the same problem. Cart can see the save files, but copynes mini does not seem to be exporting them.

Mark-VI, 2020-11-28:
All of the ROMs I've tried to dump have had 0kb save files (Zelda I and II, Dragon Warrior I, III, and IV). Also, DW III and IV won't dump correctly. Three just has a black screen when I try to load the ROM, four loads the Enix logo and then the graphics start to corrupt.

tjanas, 2020-11-28:
> All of the ROMs I've tried to dump have had 0kb save files (Zelda I and II, Dragon Warrior I, III, and IV). Also, DW III and IV won't dump correctly. Three just has a black screen when I try to load the ROM, four loads the Enix logo and then the graphics start to corrupt.

Can you share your NES rom headers for Dragon Warrior III and IV? I know that for the original Nt Mini, they require a change to the header for them to run correctly:

“Dragon Warrior III, Dragon Warrior/Quest IV, Ninjara Hoi! - Use Mapper 1, Submapper 1”

http://nerdlypleasures.blogspot.com/2017/11/analogue-nt-mini-as-nes-flashcart-and.html

Mark-VI, 2020-11-28:
> NES rom headers

This is new territory for me. Is the info you're looking for in the attached image (III on the left, IV on the right)? 

I stumbled across the text you quoted yesterday but since I am relatively new to this, I didn't know if changing the submapper is something done in the copy NES interface or if it requires editing after the dump.

[
[image]
](url)

tjanas, 2020-11-28:
Open it up in [iNES Header Editor](https://www.romhacking.net/utilities/1352/)
Or open it up in a hex editor like HxD and provide the first sixteen bytes of the file which would be the header.
Here is the special header needed for Dragon Warrior III on original Nt Mini:
`4E 45 53 1A 20 00 12 08 10 00 00 00 00 00 00 00`
[image]

[image]

Mark-VI, 2020-11-29:
Thanks for taking the time to try to help me.

I tried it both with a hex editor and with the header editor you recommended. Both produce identical results (see below), but neither work. They still just give me a black screen. 
[image]
Unless I'm missing something, they look the same as your screenshot.

tjanas, 2020-11-29:
OK, so your header with the fixes for the original Nt Mini didn't work... how about trying this fully-correct NES2.0 header for Dragon Warrior III (USA), which should work on Noir (but wouldn't work on original Nt Mini): `4E 45 53 1A 20 00 12 08 00 00 70 07 00 00 00 01`. This was generated by nes_header_repair.py

Mark-VI, 2020-11-29:
I tried the fully correct header, no luck there either. I also tried booting the ROM in an RetroArch (Nestopia) and it didn't work there. I haven't yet had a chance to try all of the variations on the header with the emulator, just the latest one.

I double and triple checked that my cart was seated correctly in the slot and running fine and I redid the dump before trying the latest header. I dumped Mega Man 5 and Metal Storm too. They work fine both in the NT and emulated.

tjanas, 2020-12-20:
I suspect something is wrong with CopyNES on Noir. I tried dumping three different Japanese Famicom carts: Super Mario Bros. 1, Super Mario Bros. 3, and Kirby. Inspecting their ROM data, for all three the PRG data matches the NES database but the CHR data does not.

Also, for Kirby, I made sure to check the battery save checkbox, and it produced a .sav file in both the CopyNES directory and in the /SAVES/NES directory, but that 8K .sav file must have been bad; I also tried making a copy of it and renaming the copy to match the good dump of the rom, and loading the game showed no save data. The save data on the Famicom cart is still intact though.

tjanas, 2020-12-20:
I dumped these on my old Nt Mini from 2017 and those dumps match the NES database and the save data was properly dumped.

neworion, 2021-01-16:
I successfully made playable backups of all of my NES carts. Each rom works, even the ones I mention below with battery saves. Yet the save files themselves appear to be corrupt. 

Dragon Warrior and Zelda 1 & 2 each used the MMC1 chip. While CopyNES created 3 *.sav files that were 8kb in size, they are not recognised by the NT mini noir or by Mesen NES emulator on the PC.

I would love to get these save files backed up.

Bohman-Timothy, 2021-04-03:
With the initial version of the Nt Mini Noir Jailbreak firmware (v6.2), I wasn't able to back up any saves either (all it gave me was a 0 KB file for each game), even though the ROMs backed up correctly, but with Nt Mini Noir Jailbreak v6.6, I successfully backed up every save file I have.

whitezombie2000, 2021-05-11:
Ok I finally got some time to try the 6.6 JB firmware. I was able to successfully dump the rom and save. The dumped rom plays fine, the dumped save loads fine, and I was able to save successfully.

neworion, 2022-02-13:
This is still a current and ongoing issue and should not be closed as Save Files do not copy over! 

I successfully made playable backups of all of my NES carts. Each rom works, even the ones I mention below with battery saves. Yet the save files themselves appear to be corrupt.

Dragon Warrior and Zelda 1 & 2 each used the MMC1 chip. While CopyNES created 3 *.sav files that were 8kb in size, they are not recognised by the NT mini noir or by Mesen NES emulator on the PC. So yes I did select the battery option while saving 😉

I tried using Jailbreak versions 6.5 & 6.6 and neither worked. I would love to get these save files backed up.

whitezombie2000, 2022-02-13:
Reopened issue.

tjanas, 2022-02-13:
I wonder when we will see version 6.7. I think it’s been at least a year since 6.6?

whitezombie2000, 2022-02-13:
@neworion 

Here is Dragon Warrior dumped by CopyNES using JB 6.6 with the NT Mini Noir. I tested the dumped ROM and save via Mesen using RetroArch 1.10.0. It worked just fine. I also tried a downloaded ROM of Dragon Warrior from a ROM site, renamed the .SAV to match the downloaded ROM, Dragon Warrior (USA).SAV,  and the save worked with the downloaded ROM via Mesen using RetroArch 1.10.0.

[GAME1117.zip](https://github.com/SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak/files/8054876/GAME1117.zip)

neworion, 2022-02-13:
I played the game you sent for a moment and then saved it. You can see that this is what the save file should look like and you can continue where I left off. I am unsure what the difference between the version you shared and the one I created that works is. All I know is the one that is dumped is not properly recognised while the one that I generated is 🤷‍♂️

[GAME1117 sav file.zip](https://github.com/SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak/files/8055723/GAME1117.sav.file.zip)

tjanas, 2022-02-13:
> If you open up the rom in the NT mini noir, an everdrive, or an emulator and then try and load the saved games you shared. You will see that it does not work and it asks you to begin a new game. This is the same error I enountered.

@neworion I just tried the GAME1117.NES and GAME1117.SAV provided by @whitezombie2000, and when I place that .SAV file in the Retroarch saves directory, with the file renamed to GAME1117.srm, when loading GAME1117.NES it does recognize the save file as legitimate.
[image]
Your GAME1115.SAV file is also good; I see it contains a save file named ADAM:
[image]

closed 2022-02-13 by tjanas

neworion, 2022-02-13:
Well, I feel like a noob now. I had the save files in the same directory as the rom every time I tried to test this out on every platform and didn't put them in the save directories. Such a basic oversight on my part 🤦‍♂️ Thanks for the help everyone

[DRAGON.zip](https://github.com/SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak/files/8056068/DRAGON.zip)