SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #12: GBC: Bear in the Big Blue House hangs after title screen

matthargett, 2020-11-25:
One of my daughter's favorite shows. We play the ROM on MiSTer fine, but it hangs after the title screen on Nt Mini Noir (and the original Nt Mini).

frederic-mahe, 2020-11-26:
According to [No-Intro](https://datomatic.no-intro.org/index.php?page=show_record&s=47&n=0109), hash values for that ROM should be:
```
881c5ea6bbbee4a29f411dc3fa9e831323ea4703b8d52107e9920228cebd42c6	Jim Henson's Bear in the Big Blue House (USA) (En,Fr,De,Es,It,Nl).gbc	e4c1c58025a083c067f752490c1fbd816c93d072	afe51a74c6967711ddd263e3998b7afd	851c8710
a4fbf540eb74be5ee2909a71f510bb25cfec3b6007f850337233eeb1add591b5	Jim Henson's Bear in the Big Blue House (Europe) (En,Fr,De,Es,It,Nl).gbc	a5c2bb5244fe0fd6d2710c832f66f8f8b8c2bbab	82880772ebd67b839c497dfe4308a0a0	48c99939
```
@matthargett could you please confirm that you have the same ROMs?

tjanas, 2020-11-26:
I just tried these two versions on my regular Nt Mini, and I can confirm I get a solid white screen after the GBC bios boot screen.

matthargett, 2020-11-26:
confirmed! and to reiterate, works fine on a MiSTer GBC core from around August and verified it still works with latest (as of November 2020) MiSTer GBC core.

matthargett, 2020-12-16:
Still broken on new 6.5 JB firmware, despite other reported bugs being fixed.

tjanas, 2021-02-23:
Please retest with JB v6.6

tjanas, 2021-12-23:
@matthargett is this still broken with JB v6.6?

matthargett, 2021-12-25:
Verified fixed with JB 6.6, thanks!

closed 2021-12-25 by matthargett