SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #13: GBC: Tigger's Hunni Safari, major graphical glitches

matthargett, 2020-11-25:
On start menu after the intro animations looks okay, persists into the game.
[image]

frederic-mahe, 2020-11-26:
Possible hash values for these ROMs are:
```
0738cbd9348166ea844bfdf76a24ea5e6bd9aebe8411e192c32b9ec40b978f15	Pooh and Tigger's Hunny Safari (USA).gbc	934918da6249723760a356748bf240f57262bd77	5f872189d58f841b5672d4cb11ec3bbb	622690da
c6de43ecf5c6fe1722b5b6d6ac375cba14c37db04d29455c295cfad4fc55b864	Pooh and Tigger's Hunny Safari (Europe) (En,Fr,De,Es,It,Nl).gbc	4141887f0b152f8592b905355e11cf43147128c2	7a58b84e406bd4df6b3c3b8807493aa8	8505d139
```
@matthargett could you please confirm that you have the same ROMs?

matthargett, 2020-11-26:
confirmed, it's the hash value for the USA one. dumped directly from my physical cartridge.

matthargett, 2020-12-16:
Fixed for me in 6.5 update.

closed 2020-12-16 by matthargett