SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #6: Some games shown as color rectangles

edrozenberg, 2020-11-25:
Some of my titles are displaying as color rectangles when I load them.

For ex. here is how **Journey to Silius** loads when interpreted into these modern cuboids :) -
[image]

Most other games load fine.

Using the latest 2020-11-23 firmware on Nt Mini Noir.

Thank you for this great project and thank you for any hints on this issue!

edrozenberg, 2020-11-25:
Was able to fix the issue by using some other version of each of the game files that didn't load properly. Also ran across some game versions which would display `Bad Header` when trying to load - was able to successfully use different versions of those also to get the title to load properly.

sociaIism, 2020-11-25:
I'm having the same issue! I'm worried as to whether I'm using the wrong ROM set or if it's a core issue. What's strange to me is that homebrew ROMs (Battle Kid pictured below) work flawlessly, but all of my "official" ROMs (SMB 2 & 3 below) are registering objects as minimal squares. It's worth noting however that games with visual issues sound perfectly normal. Regardless, a definitive solution to this would be great.
[image]
[image]
[image]

sociaIism, 2020-11-25:
I was also able to fix the issue by using a different ROM pack. That being said, I'm now getting a "Bad Header!" error for a few ROMs.

SmokeMonsterPacks, 2020-11-25:
These are very typical ROM-related issues, so I'll close this.

closed 2020-11-25 by SmokeMonsterPacks

edrozenberg, 2020-11-25:
> ...That being said, I'm now getting a "Bad Header!" error for a few ROMs.

I've seen this also. Try a few other versions of each of those ROMS. For 1 ROM I tried 3 different versions until I found one that worked.

MrUNIVAC, 2020-11-26:
I had this exact same issue on practically every other ROM with the No-Intro romset. I was able to fix it by updating all of my ROMs to NES 2.0 headers. Once I did that, all of the previously “blocky” ones worked fine!

There’s a Python script you can use on a database to automate that process. Nerdly Pleasures explains how to use it in this post: https://nerdlypleasures.blogspot.com/2020/04/fixing-nes-headers-and-renaming-nes.html.

Here’s the latest XML database:
https://forums.nesdev.com/viewtopic.php?t=19940

Good luck!

edrozenberg, 2020-11-26:
@MrUNIVAC thanks for the tips!  I suspected it was related to headers somehow but didn't have enough Retrogaming cred(its) to figure it out myself.