SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #7: README suggestion: include info on how to update firmware

edrozenberg, 2020-11-25:
I know it's 100% obvious to 99% of the users, but I didn't see any info on how to apply the JB firmware. Finally figured it out using the info from Super NT (https://github.com/SmokeMonsterPacks/Super-NT-Jailbreak).

Perhaps this info could be added to the NT Mini readme as well?

1. Copy the firmware file, for example `ntmv2_firmware_verJB6.2.bin`, to the root of the SD card
2. Insert the SD to the console and start up the console
3. Wait until the firmware update is complete. **Do not shut down the console while the LED is red. ** Wait several minutes until the process is done and the console menu appears on its own.
4. Shut down the console, remove the SD card, then delete the firmware file from the SD card using a computer
5. Follow the existing README instructions to create the directory structure on the card and add games to the directories

Dochartaigh, 2020-11-25:
This stumped me too. The directions only say:

"Unzip this pack directly to the SD card, keeping all directories/folders intact". 

Not a single file is called "pack". When you talk about a "pack", that, to me, seemed like it had to be the .ZIP file which once unzipped has 27 directories in it. So I unzipped that, put the SD card into my Noir and wondered why the LED didn't blink to indicate the firmware was being updated...

I did NOT think/know I had to even download the .bin file as nothing in the directions said to. I took a leap of faith by downloading the .BIN file to the root of the directory, then on my second try it updated correctly.

SmokeMonsterPacks, 2020-11-25:
Thank you for the suggestions, I have updated the Readme with clearer instructions. Please let me know if they need any changes or if you have recommendations. https://github.com/SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak/pull/9

closed 2020-11-25 by SmokeMonsterPacks