SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #8: Clarify support for Famicom data pad vs NTT datapad

itsumimario, 2020-11-25:
The documentation as it is currently suggests in multiple places that the famicom data pad (aka the famicom network controller HVC-051) is a viable controller choice for several of the cores that make use of the number pad. This doesn’t seem to be the case because the cores that could potentially make use of the number pad don’t read input from the famicom expansion port. So perhaps clarify that it’s the NTT Datapad SNES controller (with SNES to NES adapter) plugged into the NES port that ought to be used, rather than a famicom data pad plugged into the famicom expansion port.

frederic-mahe, 2020-11-25:
@itsumimario the README says:

> Only the NTT Data Pad or the Famicom Network Controller can fully map the numberpads of the Intellivision controller.

Do you confirm that the Famicom Network Controller (HVC-051) does not work with the Intellivision? and that only the SNES NTT Data Pad (NDK10) does?

itsumimario, 2020-11-26:
I can confirm that the HVC-051 does not work with Intellivision. I do not yet own the SNES NTT Data Pad, so I cannot confirm myself that it works, but I have heard from others that it does work.

frederic-mahe, 2020-11-26:
Thanks @itsumimario the README has been updated.

itsumimario, 2020-11-29:
Thanks for updating the documentation. I’m gonna be a tad obnoxious and point out that there still remains a reference to using the HVC-051 under the  Colecovision section of the readme that I think you forgot to strikeout or remove. :)

frederic-mahe, 2020-11-30:
you are right @itsumimario (fixed in 25f3ada4a15a0dfadae8c135b2b3dec60dca3a20).

closed 2020-11-30 by frederic-mahe