SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #14: GBC: Harvest Moon 3, major graphical glitches

matthargett, 2020-11-26:
Similar glitches to Tigger Hunni Safari, like alpha/transparency layer isn't getting cleared. Visible during character creation screen. I can provide a screenshot, if that helps.

matthargett, 2020-11-29:
Picture of the first screen after pressing Start button. This weird layer will remain for most of the new game wizard, and then turn blue for a few screens, then go away once in gameplay.

[image]

tjanas, 2020-11-29:
Just tested this on the old Nt Mini and didn’t have any issues like this, so this must be a regression.

matthargett, 2020-12-16:
Fixed for me in 6.5 update.

closed 2020-12-16 by matthargett