SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #15: GBC: Pokemon Pinball Major graphics issue

ignignol, 2020-11-27:
Title screen plus all menus display as blank screens but otherwise function properly. Game seems to display fine once you are in a table.

tjanas, 2020-11-27:
Title screen and gameplay work ok on original Nt Mini.
[image]
[image]

CheeseSodaPunch, 2020-11-29:
Here's video: https://youtu.be/a6N-NRNDxZw
SHA-256 of the rom: 7672001d4710272009df6a41e3cbada65decd56e0eb2f185cb3d59c08d33ea0e

matthargett, 2020-12-16:
Fixed for me in 6.5 update.

Elrinth, 2020-12-21:
Can original author please update if this is fixed. Please close the issue if it's okay. @ignignol

ignignol, 2020-12-21:
Yes

closed 2020-12-21 by ignignol