SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #16: More updates needed for BIOS files in README

davehensley, 2020-11-27:
Hi, following up on the BIOS file documentation issues in https://github.com/SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak/issues/2, I noticed the following:

### Intellivision:

Currently there is no CRC32 listed in the README for the BIOS file (`012.bin`), but from No-Intro it looks like it might be `0DE7579D`. Can someone please confirm?

Also, searching on Google etc. I can't find anyplace on the Internet (other than in this repo) that references the checksum listed for `intvexec1.bin`. Usually if it's the right checksum you can find a reference to it in other databases like No-Intro or MAME/MESS. The current README has it as `A85FC6DD` but No-Intro has it as `CBCE86F7`.

### Channel F

Similar to the above, I can't find a reference to the current checksum listed (`2882C02D`). No-Intro has 3 BIOS files, but I'm not sure which one (if any) is the right one to use here:
```
04694ED9	[BIOS] Fairchild Channel F (USA) (SL31253).bin
9C047BA3	[BIOS] Fairchild Channel F (USA) (SL31254).bin
015C1E38	[BIOS] Fairchild Channel F (USA) (SL90025).bin
```

### Odyssey^2

I can't find any references to the 3 checksums listed for the voice add-on BIOS files (`19355075`, `66041B03`, and `6780C7D3`) so I'm not sure they are correct.

### Sega Master System

The README currently has no checksum listed for `smsbios.bin`, but No-Intro has it as `0072ED54` for North America and Europe, and `48D44A13` for Japan.

tjanas, 2020-11-27:
Sega Master System has [many](https://datomatic.no-intro.org/index.php?page=search&op=datset&s=26&sel_s=Sega+-+Master+System+-+Mark+III&stext=BIOS&where=1&searchme=Search&pageSel=0&element=Titles&sort=Title&order=Ascending) different BIOS versions.

tjanas, 2020-11-27:
For the Intellivision speech ROM, I believe it needs the byte order reversed from the common dump (last byte first, first byte last).

tjanas, 2020-11-27:
@davehensley For the Channel F BIOS, `cfbios.bin` should be a concatenation of `[BIOS] Fairchild Channel F (USA) (SL31253).bin` + `[BIOS] Fairchild Channel F (USA) (SL31254).bin` with the checksums you listed above, resulting in a merged file with a checksum of `2882C02D`.

For example, via Windows command-line:
`copy /B sl31253.bin + sl31254.bin cfbios.bin`

tjanas, 2020-11-27:
I believe `intvexec1.bin` may need to be converted to this new INTV2 binary file format, which would explain why it wouldn't match no-intro or other commonly-found file checksums.

davehensley, 2020-11-28:
Thank you @tjanas! This is extremely helpful. I hope this information makes its way into the README to help other people, too.

Do you happen to know the correct CRC32 for the `012.bin` file (Intellivoice)?

Spazilton, 2020-12-01:
The Odyssey Bios file Checksum are correct.  These were the same bios files needed for the original NT Mini.

davehensley, 2020-12-02:
> I believe `intvexec1.bin` may need to be converted to this new INTV2 binary file format, which would explain why it wouldn't match no-intro or other commonly-found file checksums.

Thanks! I wasn't able to find a program to convert to INTV2 format, so I'm trying to build it manually. Hopefully we can document a process that others can use as well. Here's what I'm doing:

I take the common BIOS file (from no-intro, etc.):
```
$ crc32 '[BIOS] Intellivision Executive ROM (World).int' 
cbce86f7
```

Then I swap the bytes:
```
$ hexdump -v -e '1/2 "%04x"' '[BIOS] Intellivision Executive ROM (World).int' | xxd -r -p > byteswapped
$ crc32 byteswapped 
a32de182
```

The BIOS file is 8K (4K 2-byte words), and 4096 is 0x1000 (16^3), so I make the INTV2 header and footer:
```
$ echo '0010 0000 0010 0000' | xxd -r -p > header
$ echo '0000 0000 0000 0000' | xxd -r -p > footer
```

Finally I build the complete BIOS file by concatenating them all together:
```
$ cat header byteswapped footer > intvexec1.bin
$ crc32 intvexec1.bin 
eeb54c63
```

This seems to all work, but the README says the crc32 should actually be a85fc6dd. What am I doing wrong here?

Spazilton, 2020-12-02:
> Thanks! I wasn't able to find a program to convert to INTV2 format, so I'm trying to build it manually. Hopefully we can document a process that others can use as well. Here's what I'm doing:
>
> I [...]

Nothing. I used Hexdump also to byteswap the file just like you did and did the header and footers with a hexeditor. I got the BIOS working tonight and was able to boot up a intv2 format Burgertime, CRC on Intvexec1.bin on mine is EEB54C63

davehensley, 2020-12-04:
Adding to the [recent update ](https://github.com/SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak/commit/f12fdde44c5426859ceefef378183b68996d9c93) from @tjanas, a kind user on discord has confirmed that the actual crc32 for `012.bin` is actually `8bd786ec`. To get that, start with the common BIOS file (whose crc32 is `0de7579d`) and reverse the order of the bits in each byte. Here's a simple one-liner to do it on a Mac/Linux system with python:
```
cat sp0256-012.bin | python -c 'exec("import sys\nfor b in sys.stdin.read():\n sys.stdout.write(chr(int(\"{:08b}\".format(ord(b))[::-1],2)))")' > 012.bin
```

dot-bob, 2020-12-14:
> $ crc32 '[BIOS] Intellivision Executive ROM (World).int' 
> cbce86f7
> ```
> ```
> $ cat header byteswapped footer > intvexec1.bin
> $ crc32 intvexec1.bin 
> eeb54c63
> ```
> 
> This seems to all work, but the README says the crc32 should actually be a85fc6dd. What am I doing wrong here?

@davehensley I looked into this when updating my intv2 converter...  The bios with the checksum A85FC6DD is the Intellivision II bios with the extra rom that was added to try to restrict 3rd party titles.  The bios CBCE86F7 which becomes EEB54C63 is the first Intellivision executive bios without this added rom area.  There may be better compatibility using the EEB54C63 bios (I haven't had a chance to test this yet).

davehensley, 2020-12-14:
@dot-bob Ah, now it's starting to make sense. I dug into this some more and looked in Smokemonster's [SMDB file](https://github.com/SmokeMonsterPacks/EverDrive-Packs-Lists-Database/blob/master/EverDrive%20Pack%20SMDBs/Analogue%20Nt%20Mini%20%26%20Noir%20Jailbreak%20SMDB.txt) for the Nt Mini and noticed that his pack contains both versions:
```
Analogue Jailbreak/INTV/5 BIOS/Executive ROM, The (1978) (Mattel).intv   eeb54c63
Analogue Jailbreak/INTV/5 BIOS/Executive ROM, The (intv2).intv           a85fc6dd
```
So I would suggest that the README should be updated to show the crc32 of the original Mattel BIOS (eeb54c63). Does anyone disagree?

Spazilton, 2020-12-14:
> @dot-bob Ah, now it's starting to make sense. I dug into this some more and looked in Smokemonster's [SMDB [...]

I don't disagree your CRC-32 hashes match my versions. I converted both versions of the BIOS files by hand and they match the CRC-32 Checks.

dot-bob, 2020-12-16:
I say list both CRC's for both bios versions and note which belongs to the Intellivision I and Intellivision II and the user can choose.  The Intellivision I bios seems to have better compatibility with games though.  Mattel added code in the Intellivision II bios that tried to prevent 3rd party titles from working with the system.  If I use the Intellivision II bios with Donkey Kong (Coleco) I get a green screen where it works fine using the Intellivision I bios.

tjanas, 2020-12-16:
I have added additional documentation to the README with regards to both Intellivision Executive ROM source files.

closed 2023-06-26 by tjanas