SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #17: Add palette files for VS, add other custom palettes

tjanas, 2020-11-27:
At a minimum, VS Arcade palettes should be included and named appropriately in `/PALETTE/` so that they can be automatically-loaded when loading a VS rom with an NES2.0 header containing the specific palette to be used.

Also, add the additional custom palette directories to `/PALETTE/` that existed in the original Analogue NT Mini jailbreak.

GreatHierophant, 2020-11-27:
The Noir selects the correct palette automatically if it is specified in the Vs. System games' NES 2.0 header.

tjanas, 2020-11-27:
> The Noir selects the correct palette automatically if it is specified in the Vs. System games' NES 2.0 header.

We still have to add those .pal files to the directory for them to get loaded, correct?

GreatHierophant, 2020-11-28:
> > The Noir selects the correct palette automatically if it is specified in the Vs. System games' NES 2.0 header.
> 
> We still have to add those .pal files to the directory for them to get loaded, correct?

No, you do not.  You must set the correct palette by using the NES 2.0 header's variable for for that game.

frederic-mahe, 2020-11-28:
Thanks @GreatHierophant so it seems everything relies on the quality of the ROM set, and especially the correctness of ROM headers. Duly noted.

tjanas, 2020-11-28:
Should we still add extra custom palettes to provide extra options besides the built-in ones for VS and the other selectable ones in the Noir JB firmware?

GreatHierophant, 2020-12-25:
You can select any palette for Vs. games as of jailbreak v6.5, so I believe that addresses tjanas' request.  However, most Vs. Games use bizarre palettes, so any palette that would look correct on them will not on normal games.  This can be closed I think.

closed 2020-12-25 by tjanas