SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #19: Gun-Dec original cart and rom do not work properly with two controllers plugged in.

thedrdodo, 2020-12-01:
A second controller plugged in seems to trigger some kind of level select/debug mode. Game is unplayable in this state. Issue comes up with an original Gun-Dec Famicom cart, and through rom file loading. On Nt Mini Noir with 6.2 jailbreak firmware.

tjanas, 2020-12-01:
Can you run an NES controller test rom and see if it shows anything unusual?

thedrdodo, 2020-12-01:
Ran several of Nintendo's own service tool roms. No unusual issues to report. 

Issue is the same on the US version of this game, as reported here: https://github.com/SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak/issues/4

Dochartaigh, 2020-12-01:
> Issue is the same on the US version of this game, as reported here: #4

I'm who reported the US version. Thanks for posting in my error report too. What's weird on mine is another user with the same exact (MD5 matched) ROM does NOT have this issue on the USA version. As somebody who is in no way an expert on this whatsoever... but I can't help but saying how this leads me to believe it's a hardware problem that only effects certain out-of-spec consoles. The person who said his works fine has also said that his controller ports are fine and not loose. My ports are EXTREMELY loose (see the videos I posted in my topic so you can see how crazy loose they are) – 75% of original controllers are this way and they may be contributing to this issue. Can you, @thedrdodo, test how ever many original controllers/lightguns you have and let us know if they're extremely loose as well?

thedrdodo, 2020-12-01:
Hard to judge really. I don't feel like the controller ports are particularly loose. They aren't super grippy either but you still have to put a little force into it to get them out.

thedrdodo, 2020-12-01:
Wonder if the specific controller make a difference. I see that you used wireless controllers on yours in the video in your post, I used wired controllers. NES Advantage in port 1, Famicom dog bone controller in port 2.

Dochartaigh, 2020-12-01:
> Hard to judge really. I don't feel like the controller ports are particularly loose. They aren't super grippy either but you still have to put a little force into it to get them out.

Yours are WAY tighter than mine then - mine take absolutely zero force to insert or remove. And when I mean zero, it's literally zero.

> Wonder if the specific controller make a difference. I see that you used wireless controllers on yours in the video in your post, I used wired controllers. NES Advantage in port 1, Famicom dog bone controller in port 2.

If you take a look at the other video I posted on my issue (the one where I was trying out 8x different original NES controllers in mine to show how loose the vast majority of the ports are), I tried those same 5x wired controllers (including 1x Max - don't own an Advantage), and also tried 3x 8BitDo wireless receivers (and tried in nearly every combination), and as long as I have 2 of anything plugged in the game won't load so it's definitely not the type of controller.

Elrinth, 2020-12-03:
It opens some level select screen for me when launching the game. I've tried via JB and also via Everdrive N8 (old with latest firmware). Tried both Vice Project Doom and Gun-Dec...
Whenever a controller (8bitdo reciever or retrousb wireless reciever was only controllers I had) is inserted into P2, the game boots with some level select screen, where you have to select 3 times in a row. Then the game starts, but the gameplay runs superslow, but audio runs normal speed.

Connecting a Zapper to P2 instead works fine. But whenever a normal controller is in P2 slot, the game will not play normally.

thedrdodo, 2020-12-03:
Exactly the behavior I encounter too @Elrinth. Looks on every option to play this game on the system (original cart, JB, flashcart) the same thing happens with two controllers plugged in.

Spazilton, 2020-12-03:
Enable Pass through mode once the game is loaded. This corrected the issue in Vice Project Doom. However there is a bug enabling it from main menu (before loading core) that locks the system, I'm writing up a bug report for it later in the day.

Elrinth, 2020-12-03:
What does passthrough mode mean and how does it affect anything negatively and if not, then why is it not ticked 100% of the time?

tjanas, 2020-12-03:
If you enable passthrough mode (just like with the Mega Sg), it can make it more of a hassle to access the menu once in game, as the console should no longer be able to respond to hot keys. You would probably need to do a physical reset to return to the menu with passthrough mode enabled. I remember there were some [issues](https://github.com/SmokeMonsterPacks/Mega-Sg-Jailbreak/issues/6) (with controller 2 on Mega Sg that had to use passthrough mode as a workaround.

Elrinth, 2020-12-03:
Once passthru is enabled, you can't get back to the menu if you set passthru before starting a game. So first start a game, then set passthru.
After that, you can get into the menu by SELECT + DOWN (default) which takes you back to file menu if you were running a game via cores. Then you can hit start from there and you can go set it to passthru. Then you press select again to get back to the file select menu and you can re-run the game and it will be fixed.

tjanas, 2020-12-15:
Can you retest this on the latest JB firmware 6.5?

thedrdodo, 2020-12-18:
Original cartridge now plays fine with two controllers plugged in on 6.5.

closed 2020-12-18 by thedrdodo