SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #22: Genesis core button mapping problem, using M30 controller

kshark8, 2020-12-02:
According to the readme, the mapping for the Genesis core using the M30 controller should be:
Genesis | M30
-- | --
A | B
B | Y
C | A
X | X
Y | Z
Z | C

But I'm seeing:
Genesis | M30
-- | --
A | B
B | A
C | Select
X | n/a
Y | B
Z | n/a

My setup is using a NES 8bitdo BT adapter and 8bitdo M30 BT controller fot this test. I'm using Street Fighter 2 option key config to get the result I got.

Hopefully there is a way to remap the controller config for the Genesis core.

JeeseJames312, 2020-12-05:
I have a similar problem with the SN Pro 30, and in my troubleshooting I think I've identified some information which may help this issue.  I understand the readme does not talk about SN Pro 30 mappings, but I was really hoping I could use it (as well as other modern day controllers such as Switch Pro).

I found [this Analogue Support page](https://support.analogue.co/hc/en-us/articles/115001657087-Using-the-Retro-Receiver-for-NES-with-the-Nt-mini) which has a custom 8BitDo Retro Receiver firmware for the NT Mini. The download link is on this page, it's a bit difficult to locate since it does show up in typical blue highlight, but it's there.

I loaded this firmware onto my receiver, but my SN Pro 30 will not connect to them.   I believe this is because the custom firmware provided on this page is based on version 1.23, while SN30 pro support was added in version 1.27.

I synced my Wii U pro controller up to the receiver with the custom 1.23 firmware, and button mappings were great!  Select was no longer C.

I then synced my Wii U pro controller up to a receiver with the latest official 8BitDo firmware, 1.34.  The Wii U pro controller mappings changed to mimic those reported in this issue for the M30, which reflect the mappings for the original NES controller.

I believe folks experiencing this problem with the M30 can resolve the button mapping issue by using the Custom Analogue 1.23 firmware.  I believe a new custom firmware will need to be created based off the 1.34 official 8BitDo firmware to support better controller mappings for any new controller added to the Retro Receiver firmware since 1.23, such as the SN30 Pro, the Xbox Elite 2 controller, and Switch Pro Controller.

@kshark8 , I do not own a M30.  Could you try this custom 1.23 firmware on your retro receiver and report back if that fixes the M30 mapping?

kshark8, 2020-12-05:
> I have a similar problem with the SN Pro 30, and in my troubleshooting I think I've identified some information which may help this issue. I understand the readme does not talk about SN Pro 30 [...]

Cool, I downloaded the FW u recommended, i will try the moment i get home and let you know if it works or not. Thanks alot!

kshark8, 2020-12-06:
> @kshark8 , I do not own a M30. Could you try this custom 1.23 firmware on your retro receiver and report back if that fixes the M30 mapping?

I finished testing witht the custom RR 1.23 fw, unfortunately it is not compatible with the M30 BT controller. It doesn't connect at all, I tried all the modes, only the mac mode connects to the RR but none of the input buttons work. The RR 1.23 fw does work for the oldest 8bitdo NES and Famicom controller that they released for the NES mini and famicom mini. In this case, the mapping is as such:

Genesis | 8bitdo NES controller
-- | --
A | B
B | Y
C | A
X | X
Y | LS
Z | RS

It's much better than before, thanks for your help @JeeseJames312 . But I really hope analogue release a new custom RR fw that would suppose the new controllers such as M30 and SN 30 pro and Pro+.  Especially the M30 because it's a complete duplicate of the Genesis controller which makes most games play identical. Having the Genesis ABC mapped to Nintendo BYA, makes it awkward to use because it would feel like BAC on a Genesis if you know what I mean.

dot-bob, 2020-12-10:
My guess the custom firmware runs the NES receiver in the SNES mode maybe with slightly different button mapping.
If you use a SNES retro receiver with a SNES to NES cable you can use the M30 with the following mapping:

Genesis | M30
-- | --
A | A
B | X
C | B
X | Y
Y | Z
Z | C

Still not great but is at least playable unlike with the NES retro receiver with select being C.

Dochartaigh, 2020-12-12:
Please excuse me for butting in here (or please break off this into another topic if you deem it mods), but I just wanted to add this is more a problem with 8BitDo's HORRIBLE firmware for all their their wireless products which are used on the Nt mini Noir (and I say that with a heavy heart, I am a fan and own TWELVE 8BitDo controllers, 11x with Retro Receivers: 3x N30 2.4g, 2x SN30 2.4g, 2x NES30 BT/Bluetooth, 2x SNES30 BT, 2x M30 2.4g, M30 BT and I'm probably forgetting a couple... ;)

In regards to using 8BitDo controllers with the Nt mini Noir, with the new 2.4g NES Retro Receiver or the original Bluetooth NES Retro Receiver (which is needed to run other BT non-NES style controllers on a NES), ––everything on 8BitDo's newest firmware for both the receivers and controllers–– ALL the button mappings for every single one of these 8BitDo controllers –except the N30 2.4g type the NOIR came with– are messed-up, and also missing turbo. 

These will have either B and A position switched (doesn't sound like a HUGE problem, right? It IS!!! - try playing SMB where you have to hit B with the tip of your thumb, then feather down to hit A at the same time with the middle part of your thumb - this action is required to run and jump at the same time). 8BitDo also messed-up Turbo on a bunch of these controllers when used on the Noir. I can get it to work on ONE single model - the SNES30 BT, and that's only if you roll back the firmware!!! (which then gives you connect-from-wake issues and you have to commonly re-pair a controller than was paired properly before...).

I can go on and on, and if any of the developers want to reach out for me to do some testing please let me know - I have ever retro style 8BitDo + receivers + a Noir + original console hardware if needed (multiples of NES/SNES/Genesis) to help. Before you ask 8BitDo customer support, although friendly and trying to help, are completely hopeless. I have two different email strings with them (on TWO different emails because they kinda stopped answering my one lol... I was being nice, don't worry ;) where I'm literally sending them screenshots from their OWN manuals, telling them my X, Y, and Z controllers do NOT work that way... and they keep on telling me it's normal lol (which makes no sense). They also say their 2.4g models are NOT compatible with each others Retro Receivers (example: using SN30 2.4g and M30 2.4g on the NES 2.4g Retro Receiver), although these DO in fact work - the mapping is just absolutely horrible.

Somebody from Analogue, who I assume has a contact they work with at 8BitDo, really needs to use their influence to get this sorted. This would especially be a help for (older) people like me who physically have pain when they use the super small NES controller. My only way to comfortably play (after the Switch Pro kept dropping presses) is with a BT SNES30 BT where I rolled back the firmware (and won't pair like I mentioned above - on both of them).

tjanas, 2020-12-12:
Chris Taber, the CEO of Analogue, is also the Chief Marketing Officer at 8bitdo. FYI.

Dochartaigh, 2020-12-12:
> Chris Taber, the CEO of Analogue, is also the Chief Marketing Officer at 8bitdo. FYI.

Thanks, I did not know that. They should DEFINITELY have some pull to get this sorted then! It's really quite messed-up. ...like why have all these retro receivers which can use different controllers on them... then not have a single one work properly (especially when it's even shown in their manuals how they *should* work...yet still don't lol).

ferllen, 2020-12-15:
For me, M30 and 8bitdo Arcade Stick don't work properly when the retro receiver is updated to the latest 1.36 firmware version.
It's recognized as an NES controller (using select as a regular button). The problem is all my analog controllers are updated and they don't work with the retro receiver in the firmware available in the analogue's page.

ktuluboy, 2021-01-10:
I just got an 8bitdo M30 2.4g expecting to be able to have the exact Genesis 6-button mapping.  Unfortunately, it seems to have the same behavior than the BT version mentioned previously.  In fact, it seems to behave like a NES controller... where there's only 3 functional button...and the C being on the Select instead of the C of the M30.  Before I return the controller (it is unusable the way it is), is there any plan to fix this?

PS: I can help you guys troubleshoot if you want me to test special firmware/genesis core version with the M30.  I think the Genesis core is one of the coolest core of the Jailbreak and it is a shame we cannot really use it right now.

zentis, 2021-01-17:
I have the same issue with the M30 BT and the retro receiver. If I use the last firmware for the retro receiver, I can connect my M30 but X and Y are unavailable and C is mapped to select which makes the genesis core pretty much unplayable.

I've used the custom firmware from Analogue for the retro receiver but then I can't connect the M30 no matter the mode. I tried downgrading the M30 to 1.10 but that didn't solve the issue.

I've send an email to Analogue asking them if they plan to update the modified firmware for the retro receiver. I wish all of this were opensource so that I could just take action and fix it myself, it's so frustrating not being able to do anything.

zentis, 2021-01-25:
Reply to my email:

> I have passed a message detailing this issue to our development team. Be sure to follow our firmware updates for a resolution to the problem. Please let me know if you have any additional information or questions concerning the issue.

So I'm hopeful that Analogue will update the firmware from the retro receiver

kshark8, 2021-01-25:
I'm surprised they got back to you, great work. Hopefully they can provide us with a new firmware update for the console. Finger crossed!

prestia, 2021-04-26:
I reached out to Analogue about this too and got a much more terse, "I will pass along your feedback to our development team." Here's hoping something happens. I'd love some open-source FW for the retro receiver.

tjanas, 2021-04-26:
I've gotten similar emails from Analogue back in 2017 related to bugs on the original Nt Mini, and nothing ever came of it.

zentis, 2021-05-31:
Yes, no response since the last email. I'm seriously annoyed at Analogue now. Regret buying such an expensive console from a company that cares so little about their customers.

kshark8, 2021-05-31:
> Yes, no response since the last email. I'm seriously annoyed at Analogue now. Regret buying such an expensive console from a company that cares so little about their customers.
>

Ya i feel the same way. Therefore, I ended up dumping my money into MiSTer and didn't pick up Analog's MD and SNES FPGA consoles. On mister, all the controller are customizable, and cores and frequently updated.

thomaskaelin, 2021-07-08:
I'm facing the same issue with my M30 BT and NES Retro Receiver, so I decided to do some experiments on my own. I actually got the M30 working **once** with the following button mappings:

| Genesis | M30 |
| ------- | --- |
| `A`     | `A` |
| `B`     | `C` |
| `C`     | `Z` |
| `X`     | `B` |
| `Y`     | `X` |
| `Z`     | `Y` |

Here is what I did:

1. On the Retro Receiver, I installed the [custom firmware by Analogue](https://support.analogue.co/hc/en-us/articles/115001657087-Using-the-Retro-Receiver-for-NES-with-the-Nt-mini) based on `v1.23`
2. On the M30, I installed the [initial firmware release](http://download.8bitdo.com/Firmware/Controller/M30/) `v1.00`. Please note that this firmware is not visible in the [upgrade tool](https://support.8bitdo.com/firmware-updater.html) provided by 8BitDo. But you can trick the upgrade tool to install that firmware: just install any other firmware on your M30. After completion, you will find a `.dat`-file inside the `updateFile`-folder (a local cache folder used for storing the downloaded firmwares). Simply replace that file with the `.dat`-file from the manually downloaded `v1.00`-release and let the upgrade tool install the exact same version as before. The tool won't verify any checksums of the firmware before installing, it simply uses any file having the proper file name from the cache (eeeks!!). However, the tool will install the `v1.00` on your M30.
3. After all these shenanigans, you should be able to pair your M30 with the RR using the `macOS`-mode. It took my some tries, but eventually the controller was paired and working (with the button mapping shown above).

As I already mentioned, I was only able to pair the devices successfully **once**. No idea what I did to make it work, honestly. I'm hoping my findings will help anyone else to get this working or make at least some more progress. Would really appreciate to use my M30 for the Genesis-core.

thomaskaelin, 2021-07-09:
@kshark8 I made some more experiments today. I am now able to repeatedly pair my RR and M30 support all six buttons. Here is what I did:

1. On the Retro Receiver, I installed the custom firmware by Analogue (`v1.23`).
2. On the M30, I installed the most recent firmware (`v1.14`). 
3. I switched the M30 to `macOS`-Mode.
4. To pair the RR with the M30, I used a USB-cable (USB-C to USB-Micro)
5. After some seconds, RR and M30 are paired. You can now remove the cable - the wireless connection will take over.

So after this success I did some experiments using the integrated *Controller Test Tool*:

| M30 | Test Tool | Genesis Core |
| --- | --- | --- |
| `A` | `1P-A` | `A` |
| `B` | `3P-A` | `C` |
| `C` | `3P-Start` | `Z` |
| `X` | `1P-B` | `B` |
| `Y` | `3P-B` | `X` |
| `Z` | `3P-Select` | `Y` |
| `L` | `3P-Select` | `Y` |
| `R` | `3P-Start` | `Z` |
| `Start` | `1P-Start` | `Start` |
| `-` | `1P-Select` | `Select` |

That still doesn't match the table in the `README`. I guess that table was never correct. Assuming that the Genesis Core reads the signals similar to the test tool, adopting the mapping like this should make the controller finally work properly:

| Genesis Core | Test Tool |
| --- | --- |
| `A` | `1P-A` |
| `B` | `3P-A` |
| `C` | `3P-Start` |
| `X` | `1P-B` |
| `Y` | `3P-B` |
| `Z` | `3P-Select` |
| `Start` | `1P-Start` |
| `Select` | `1P-Select` |

Hope that helps.

kshark8, 2021-07-09:
> So after this success I did some experiments using the integrated _Controller Test Tool_:
>
> M30	Test Tool	Genesis Core
> `A`	`1P-A`	`A`
> `B`	`3P-A`	`C`
> `C`	`3P-Start`	`Z`
> `X`	`1P-B`	`B`
> `Y`	`3P-B`	`X` [...]

Thanks for testing. I can follow everything for the instruction for connecting the RR custom FW to the M30 v1.14. But I'm confused in the last part of your reply. What is the integrated control test tool you mention? I'm not aware of a test tool for the M30, or is that a test tool for the mini noir?

Also i don't understand the last table, how do you go from the first table to the last table.  In other words, how were u able to match the M30 button map to the M30 core button map? Because your test result is first table and last table result doesn't match.
 
Sorry maybe I missed something but i read your findings several times, but not sure what you did in test tool to match the M30 to m30 core, because I don't know what the test tool does and where to access it.

thomaskaelin, 2021-07-09:
@kshark8 In the main menu of the Nt Mini Noir, there is a section called **Tools**. In there you find an option **Controller Test** which let you test your controller. Whenever you press a button on your controller an icon is highlighted corresponding to the data signal the console received from the controller.

The first table contains my observations with the mentioned configuration. The second table contains to the *expected* mappings for the Genesis Core as an input for the developer(s) of the jailbreak. As an example, the core should map a `1P-A` data signal as a press of the `A` button, etc.

Hope that made things more clear. :-)

kshark8, 2021-07-09:
I see, you mean the last table is the ideal button mapping for the jailbreak developer, I misunderstood and I thought you for it working as in last table. 

Ya I agree, i believe the the jailbreak genesis core using the M30 is never accurately documented in the jailbreak help file. This is why I created this thread to address the problem, hoping the jailbreak author can update and address the issue. No solution it seems so far from the jailbreak developer and 8bitdo. 

Ps: The genesis core seems to be limited to design for the usual 4 button controller with LR, like the SNES controller.

eltees, 2021-07-11:
FYI while it won't help with the M30 directly unfortunately, using a snes to nes cable, a snes retro receiver on current firmware, and an 8bitdo pro 2, you can use their input customizing apps to actually get 'normal' button layouts you want.  (well as normal as any snes style controller is feeding genesis 6 button input).   Its an awful lot of hardware to throw at getting the core playing 'normally' and its unfortunately not as nice as using an m30 directly would be, but you can get simple/effective face button A-B-C left to right and add in the left/right triggers alone (L1, R1, R2 as x y z, or blending the X face button for L1, X, R1) etc.

Really the best solution(s) would be doing another revision of the analogue custom firmware, but based on the current retro receiver core and the modern controllers for it, or some way to toggle in the hardware features to treat the inputs as if they were snes style inputs allowing the retro receiver on stock firmware to work as a snes style input.

Unfortunately both would require custom work from analogue or the jailbreak, and both would still strongly benefit from allowing/supporting a per-core input remap file of some kind to allow people to use their controllers as they please.

VinceVaughn, 2021-07-12:
A Retro Receiver, with the custom firmware by Analogue (v1.23), can easily pair with a Wii U Pro controller, if you have one, even on a NT Mini Noir.

It's mapped this way :

Wii U Pro        Genesis Core

+ start                   Start
    Y                           B
    B                           A
    A                           C

Wii U Pro controller's buttons are Y-B-A from left to right, so you get a "B-A-C" configuration on the Genesis Core.
Not perfect, but in most games options, you can easily change the buttons configuration to get closer to the original feeling.

Not sure to keep the two M30 I bought for the Genesis core...

eltees, 2021-07-12:
> A Retro Receiver, with the custom firmware by Analogue (v1.23), can easily pair with a Wii U Pro controller, if you have one, even on a NT Mini Noir.
>
> It's mapped this way :
>
> Wii U Pro Genesis Core [...]

Other functioning controllers for that special firwmare I've found are original 8bitdo n30 pros (the original tiny ones that have glowing sides and micro-usb charging, not the current ones that use usb-c charging), Wii remotes (haven't tried with the classic pad that you would need to get sufficient buttons), ps3 controllers, some early bluetooth xbox pads, some ps4 controllers (might vary based on the pad build/firmware versions for those two).  

Unfortunately pretty much everything made by 8bitdo in 2018 or later (almost 1:1 with charging via usb-c vs micro-usb) will not pair with it successfully since they will not be in the retro receiver 1.23 whitelist to connect to.

closed 2023-06-26 by tjanas