SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #25: Zelda won't load .sav

phillipc3, 2020-12-02:
I've got the most recent 'jailbreak', I've tried dumping the save file off my Zelda cartridge and it came out with a size of 0. Then the save disappeared off my cartridge (the battery died, it was on its way out anyway). Used a save file editor to edit a .sav I had on my PC to get myself back to where I initially was, put it in the saves/nes folder, and when I boot up the game from cart or SD card there are no saves. 

Every time I try and dump saves off the cartridge, no matter the amount of progress, the size is 0.

Am I doing something wrong?
[GAME1115.zip](https://github.com/SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak/files/5632662/GAME1115.zip)

Elrinth, 2020-12-04:
Did you replace the battery?

phillipc3, 2020-12-04:
> Did you replace the battery?

Yes, brand new in a battery holder (I normally use these as you can preserve save files when changing battery). Dumping using MMC1, I checked some more games and none save onto the SD

tjanas, 2020-12-21:
Something is wrong with CopyNES on the Noir, the actual ROM dumps seem to be incorrect and those that have SRAM aren’t dumped properly either (when the battery checkbox is checked). The save file I dumped from my Japanese Kirby cartridge did not dump correctly (but copying the save file I dumped with my original Nt Mini worked fine).

tjanas, 2021-02-25:
Duplicate of https://github.com/SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak/issues/11

whitezombie2000, 2022-02-13:
This issue should be closed.

closed 2022-02-14 by tjanas