SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #26: Famicom Keyboard with Family Basic input wrong keys

Elrinth, 2020-12-03:
When using the famicom keyboard and the original cartridge Family Basic with a normal Famicom. The keyboard inputs the keys you press.

But when pressing keys on the Analogue Nt Mini Noir with jb (didn't try without) it will input keys a bit more to the right. Example: I press A, however it inputs into the game a D. No key is correctly mapped.

Any clue why this happens?

tjanas, 2020-12-03:
Did you enable controller passthrough mode? It is required to use the Famicom Keyboard.

Elrinth, 2020-12-03:
You are correct, this is a non-issue once enabling key passthrough it worked! Didn't even have to reset the game.
Sad we lose the ability to get into the menues after doing that.

frederic-mahe, 2020-12-04:
documentation fixed with 71f9982514070d614fd8ceb6a3b2c87a3848d3ec

Elrinth, 2020-12-21:
Closing issue as there is a workaround and reason for how it works. Thank you frederic-mahe

closed 2020-12-21 by Elrinth