SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #27: Kool aid man

ralph1011, 2020-12-03:
Glitched screen, can’t start game

tjanas, 2020-12-03:
What are your file checksums?

ralph1011, 2020-12-04:
I did a quick look and it all seems to match no intro from the link you provided. I’ll post it shortly.

ralph1011, 2020-12-04:
CRC32 (No Intro): D9BD009B
CRC32 (Mine): d9bd009b

MD5 (No Intro): 534E23210DD1993C828D944C6AC4D9FB
MD5 (Mine): 534e23210dd1993c828d944c6ac4d9fb

SHA-1 (No Intro): 2F550743E237F6DC8C75C389A01B02E9A396FDAD
SHA-1 (Mine): 2f550743e237f6dc8c75c389a01b02e9a396fdad

GreatHierophant, 2020-12-04:
Not a bug with the Nt Mini, Kool-Aid Man uses the Difficulty A switch to pause the game.  Make sure Difficulty A is checked, the game will start.

tjanas, 2020-12-04:
Having the 2600 switches as core menu options is a nice feature of the Noir JB. With the original Nt Mini JB, you needed a controller with **L** and **R** shoulder buttons to toggle the switches.

closed 2020-12-04 by frederic-mahe

ralph1011, 2020-12-04:
> Closed #27 via #31.

Def a nice feature, thanks for highlighting it...and providing some hand holding. Much appreciated.

Just wondering, is there any tutorial anywhere about changing intellivison roms to the .intv2 format? It wasn’t all that clear in the faq. That or I am a complete novice and it went over my head. 

Thanks again