SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #32: NES: Battletoads Double Dragon Crashes

arekuse, 2020-12-05:
Both NTSC and PAL version crash upon reaching the first boss and him starting to break the gate.
This issue is probably because the game is trying to read some memory location it actually don't use.

Setting all FF or all 00 in memory before loading didn't help.

Either the game just freezes and the audio continues or, like in the PAL version, the game crashes completely.

GreatHierophant, 2020-12-06:
This crash can be avoided by using a NES 2.0 ROM header with this game.  The reason is that if the iNES 1.0 ROM header is used, the system will assume the ROM has 8KiB of WRAM at $6000-$7FFF.  This is fairly typical with emulators, but they often will whitelist known games with this issue like Battletoads and Double Dragon and Low-G-Man.  Whitelisting is a bad habit and not friendly to FPGA solutions as it requires calculating and comparing a game's checksum to a table of problematic games.  A NES 2.0 ROM instructs the jailbreak not to assume anything about the ROM except what is required by the mapper and indicated in the other fields in the header.  In these cases, the NES 2.0 ROM will indicate no extra WRAM is present, and these games will rely on the open bus behavior which worked for them as physical cartridges.

arekuse, 2020-12-07:
Hello, thank you! Although I updated headers using the python script, but I need to check once more, if it has correctly updated this game.

arekuse, 2020-12-07:
The ROM is not in NES 2.0 format, so it needs to be updated.

arekuse, 2020-12-07:
I confirm, that correcting the header also corrects the game. I close the issue.

closed 2020-12-07 by arekuse