SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #34: 7800 core

acidrain97, 2020-12-07:
I'm guess this has to do with the POKEY chip not hearing any music played in Commando. SFXs are there just no music.

tjanas, 2020-12-07:
> I'm guess this has to do with the POKEY chip not hearing any music played in Commando. SFXs are there just no music.

Did you verify that byte 0x36 is set to a value of 0x03 in your Commando rom?

acidrain97, 2020-12-07:
How is that done and how do I change it.

tjanas, 2020-12-07:
> How is that done and how do I change it.

I recommend [HxD Hex Editor](https://mh-nexus.de/en/hxd/)

acidrain97, 2020-12-07:
Looking at this page
http://7800.8bitdev.org/index.php/A78_Header_Specification#Adding_Or_Modifying_an_A78_Header_on_an_Existing_ROM

I simply do not understand where to find byte 0x36 or how to change it. I have been googling but I guess this stuff is beyond my comprehension. I've looked into 7800 header editors can't figure that out either. So I give up.

Spazilton, 2020-12-14:
Here is the byte that needs to be changed. You must open it in a hex-editor.

[https://imgur.com/a/bzYgM9G](url)

acidrain97, 2020-12-15:
got it...  thanx

closed 2020-12-15 by acidrain97