SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #41: NES: Startropics - Won't register name

Elrinth, 2020-12-12:
Startropics via the NT Mini Noir jailbreak NES core won't register name. Once you input the name it still shown as blank afterwards.

Via Everdrive N8 u can play the European version and it works fine.

tjanas, 2020-12-12:
A few things... where/how are you getting these checksums? If you open your "StarTropics (USA).nes" rom in [NESExplorer](https://www.romhacking.net/utilities/1425/), it should report that the PRG+CHR checksum (that's the checksum of the rom data, meaning that it's not including the header in its checksum calculation) is `889129cb`, which would match the following:

```
	<game><!-- Licensed North America\Startropics.nes -->
		<prgrom size="262144" crc32="BEB88304" sha1="224D96BF55391C7E9B651EB4D6A8F06FAF6E29DA" sum16="B119" />
		<chrrom size="262144" crc32="42236F71" sha1="A6038C0785BE19C1215351E5F053272BF231B78F" sum16="A9C5" />
		<rom size="524288" crc32="889129CB" sha1="74C53FE9AC779F146C59AC01E701C9BF912B3C7B" />
		<prgnvram size="1024" />
		<pcb mapper="4" submapper="1" mirroring="H" battery="1" />
		<console type="0" region="0" />
		<expansion type="1" />
	</game>
```

With a correct NES2.0 header applied, you could expect the checksum of the file to be `5F212B6F`.
Note that checksums are case-insensitive.

The header bytes for the rom referenced above should be the following (and as generated by the NES2.0 header script):
`4E 45 53 1A 10 20 42 08 10 00 40 00 00 00 00 01`

tjanas, 2020-12-12:
If you ran the script, are you sure you set `TRIAL_RUN = 0` in the python script? I would also recommend setting `VERBOSITY = 4`. The script will print out the old and new NES2.0 header if it was incorrect..

tjanas, 2020-12-12:
[image]

tjanas, 2020-12-12:
In regards to checksum `2ED79B73` that you mentioned initially, I suspect you are using this file and that you did NOT let the script alter its header, because this SMDB entry has an iNES1.0 header and is technically incomplete/incorrect for old Everdrive N8 compatibility:
`EverDrive N8/1 US - R-Z/StarTropics (USA).nes	fdd3d68c7b5b2f9dc7ec50d9888b0a8edbaaec28	02c6bc59abdf527679a7962c5b390ea6	2ed79b73`

Once you run the script correctly it should change this file:
```
X:\tmp\20201212\test>nes_header_repair.py
Reading XML, please wait...
Evaluating files...
Updating header for SHA1:74C53FE9AC779F146C59AC01E701C9BF912B3C7B File: StarTropics (USA).nes
4e 45 53 1a 10 20 42 00 00 00 00 00 00 00 00 00 (old)
4e 45 53 1a 10 20 42 08 10 00 40 00 00 00 00 01 (new)

Processing complete.
```

Elrinth, 2020-12-12:
Thank you guys! It is indeed working as intended now. This can be closed now.

closed 2020-12-12 by Elrinth

sponge9012, 2020-12-17:
i am still having trouble with this. startropics 1 and 2 can not add a save file name or progress beyond that screen

tjanas, 2020-12-17:
> i am still having trouble with this. startropics 1 and 2 can not add a save file name or progress beyond that screen

@sponge9012 Have you confirmed your StarTropics (USA) nes rom is headered correctly? You should expect the file to have a CRC32 checksum of `5F212B6F`

sponge9012, 2020-12-18:
i apologize, i did not, i have a couple of everdrive packs as well as other sets and none of them work, but none of them are that CRC. strange as one set is particularly recent. what CRC am i looking for to get startropics 2 working, as it has the same issue?

sponge9012, 2020-12-18:
if i am to understand correctly, the reason this happens is due to a CRC check being needed to switch to the MM6 mapper, so a modified rom wont run. would that be considered a bug, if a modified rom runs on an everdrive / original hardware ? i would like to use the music fix alteration, as well as the movement fix hack. is there a way to force the MM6 mapper on a given rom? or a better way to soft patch in these modifications ?

sponge9012, 2020-12-21:
> > i am still having trouble with this. startropics 1 and 2 can not add a save file name or progress beyond that screen
> 
> @sponge9012 Have you confirmed your StarTropics (USA) nes rom is headered correctly? You should expect the file to have a CRC32 checksum of `5F212B6F`

thank you, not only did it work, but it seems that i am also able to add IPS patches and it still works. thanks