SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #45: GB & GBC cores have squished picture on PVM

leavert, 2020-12-15:
The digital output is great, perfect and wonderful. BUT unfortunately when I hook the NTMiniNoir to my PVM-20M2MDU the picture is vertically squished. I do not have this problem with the NES core or reading the cartridge. In comparison my GameCube with GBPlayer has a perfect aspect ratio on the PVM leading me to believe the issue is with some setting I'm unaware of or the core.

leavert, 2020-12-15:
[image]
Gameboy Player
[image]
NY Mini Noir

leavert, 2020-12-15:
And now that I'm looking at the pics side to side I'm realizing that the image is actually stretched horizontally not squished vertically.

Firebrandx, 2020-12-22:
I assume you made sure to set the image to exact integers on the horizontal as well as vertical? If so, it may be a problem with how much of a porch is used compared to the GBPlayer. Also you should edit the tile of this report to reflect that it's an issue with horizontal stretching rather than vertical.

leavert, 2020-12-22:
All changed! I will check the exact integer setting when I get home!

leavert, 2020-12-22:
Ok I'm a bit confused. What settings allow you to "set the image to exact integers on the horizontal as well as vertical"? I don't see those settings for the analog output.

tjanas, 2020-12-23:
I don’t think adjustments apply to analog video output.

Firebrandx, 2020-12-23:
Ah okay my apologies. I  wasn't sure if analog output had the same options or not to adjust horizontal size. If it doesn't, then yeah, that's a shame because you won't be able to dial in handheld display aspect ratio.

GreatHierophant, 2020-12-25:
The GB and GBC Aspect Ratio in analog mode is aspect ratio that the Super Game Boy produces, so it is not without precedent.

Firebrandx, 2020-12-25:
> 
> 
> The GB and GBC Aspect Ratio in analog mode is aspect ratio that the Super Game Boy produces, so it is not without precedent.

While that's true, it would be nice to have the option of square pixels if possible, since that's handheld DAR.