SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #46: famicom ram adapter

acidrain97, 2020-12-15:
Could be me or maybe a real problem. First off I know once jailbroken what the point of it but....  Games like Super Mario, Zelda, are a ransom graphic mess. Some games like .vs Excitebike, SMB 2 runs fine. When running on the Analogue RAM cart diags (select+start before the boot screen) should say CRAM PRAM OK, instead some random hex code. Also expanded audio is not working, ie Zelda 1. Real disks/disksystem/fcsstick are used for testing. Famicom carts seem to run ok. 

Running the current jailbroken firmware.

The cart connector is clean as a whistle. Why some games work fine others do not. I have 2 ram carts for testing both are version 1.

thx

tjanas, 2020-12-15:
If you are using a real physical FDS RAM Adapter:
1) did you enable Fully-Buffered mode (assuming your HDMI display is being used)? Nt Noir does NOT support real FDS RAM Adapter with HDMI in zero-delay mode.
2) did you enable cart audio?

Are you seeing a hex code during FDS RAM Adapter diagnostics like shown [here](https://tcrf.net/Family_Computer_Disk_System)?

I have used my real FDS RAM Adapter with Noir in fully-buffered mode (or when only using a CRT) and I was able to do a full playthru of SMB1 without issue.

acidrain97, 2020-12-15:
First I want to address the video problem:

Using HDMI
Running on a REAL FDS Ram Adapter
Fully Buffered - Default

Some random error codes that show up 
0000H 
00AAH
0074H

Again 6.2 exhibits none of these problems. Everything runs as it should. 6.5 has some kind of issue. Both RAM adapters do the same thing. Random error codes. AVS, Famicom, firmware 6.2 work perfect.

I don't have a compatible CRT to test. 

Any other suggestions?

tjanas, 2020-12-15:
I will give my real FDS Adapter a try and report back. If it worked on 6.2, maybe something regressed

acidrain97, 2020-12-15:
thanx I appreciate it...

ZReport, 2020-12-16:
I can confirm this issue as well.  The error codes:

013AH
007AH

My RAM Adapters are just fine.  Here is a video just showing that I'm running HDMI Fully Buffered & JB 6.5 with the issue:

https://www.youtube.com/watch?v=x6AdHcGhqKU

Elrinth, 2020-12-17:
I have also random hex number showing up.,

sponge9012, 2020-12-17:
my ram adapter worked fine in 6.2 and now it doesnt. i will attempt to mess with the suggested settings but i did not have to adjust anything before.

Peredonov, 2020-12-17:
Confirming the above on the current non-JB firmware 1.1 also, via analogue video output to a CRT. Graphical issues are also immediately apparent on Super Mario Bros since the title screen.
CRAM test gives errors. With the same RAM Adapter, test gives CRAM OK result on a real Famicom.

sponge9012, 2020-12-18:
my ram adapter with fdstick is running and loading games fine. i tried mario, zelda, doki panic, metroid. i was able to get expansion audio to work in zelda by going to audio settings and turning on the FDS chip. 

however, a ram test does not check out as OK, i get a different ram address every time i do the check. I can do the ram test through the NES core and it checks as OK.

i havent performed the ram test on other firmwares, so i have no true basis for comparison, as i have lost no functionality that i am aware of.

Peredonov, 2020-12-18:
> i was able to get expansion audio to work in zelda by going to audio settings and turning on the FDS chip.

To get the real audio from the RAM Adapter you'd want to enable "cartridge audio" instead of the simulated expansion chip like this. That said, on my tests there were no problems with the real expansion audio from the RAM Adapter.

acidrain97, 2020-12-18:
> my ram adapter with fdstick is running and loading games fine. i tried mario, zelda, doki panic, metroid. i was able to get expansion audio to work in zelda by going to audio settings and turning on [...]

This may matter but do you know what version of the ram adapter you're on? If your ram adapter is working try hitting the reset a few times and what happens. Mine makes the screen worse. Some games work fine others do not. 

I can confirm running the ram tests will show OK if run though the NES core.

Also is it possible to rom dump a ram adapter's bios through copynes? if so how?

I've switched back 6.2 for the time being. At least the FDS works on that firmware.

sponge9012, 2020-12-19:
> This may matter but do you know what version of the ram adapter you're on? If your ram adapter is working try hitting the reset a few times and what happens. Mine makes the screen worse. 

no idea what version, but resetting does not make it malfunction. mine is working fine as far as i can tell. 

i did more testing. this time i reverted back to 6.2 firmware. ram check gives me the OK. then i went back to 6.5 and i get the ram address errors.

tjanas, 2020-12-22:
I can confirm this issue too. Definitely a bug.

tjanas, 2021-02-22:
This is still broken in the latest official firmware v1.2.

Peredonov, 2021-02-22:
> This is still broken in the latest official firmware v1.2.

In some ways it may be even more broken. Zelda 1 is totally bugged whereas before it loaded seemingly without issues in spite of the RAM test fail.

tjanas, 2021-02-22:
@Peredonov can you create a new issue with this Zelda 1 FDS bug? So that it doesn’t get lost in this RAM Adapter test fail bug?

Peredonov, 2021-02-22:
Technically there is no jailbreak firmware yet with the changes on official 1.2 FW, so I'm not sure if I can create an issue about it here. For reference though, here is a picture of one of a few manifestations of the problem in Zelda 1 FDS: https://media.discordapp.net/attachments/446407171471835136/813518219573723136/IMG_7682.JPG

tjanas, 2021-02-22:
Please create a new issue. It’s ok that it’s for official firmware, just note it in the details.

Peredonov, 2021-03-07:
See my last comment on #95  for info on a workaround (turning dejitter on). I suggest this issue remains open since dejitter off being linked to this problem is still an unsolved regression.

tjanas, 2023-05-04:
Duplicate of https://github.com/SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak/issues/95
Please re-test with latest official firmware https://www.analogue.co/support/nt-mini-noir/firmware/1.3

closed 2023-05-04 by tjanas