SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #52: GBC: Super Mario Bros Deluxe pallete issues

superretro2741, 2020-12-16:
[image]

Elrinth, 2020-12-17:
supposed to look something like this:
[image]

superretro2741, 2020-12-17:
Yeah the entire games colour is way off for some reason

tjanas, 2021-02-23:
Please retest with JB v6.6

superretro2741, 2021-02-23:
retested with Jb v6.6 it doesn't appear to be fixed.

tjanas, 2021-02-23:
To confirm, you updated both the firmware and the core files (those from the SD Card System Structure)?

superretro2741, 2021-02-23:
I wasn't aware the cores had to be updated like that with new files from the file structure, this has fixed the issue and it works perfect now.

tjanas, 2021-02-23:
Yep it is in the README

tjanas, 2021-02-23:
Please close the issue if it is fixed

closed 2021-02-23 by superretro2741