SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #56: Turbo Blaster compatibility

treismac, 2020-12-17:
I apologize ahead of time for posting this Analogue Nt Mini Noir compatibility issue here if it is inappropriate, but I was directed here in a Facebook Analogue group.  The Turbo Blaster peripheral does not work for the Noir.  Controllers are completely unresponsive when plugged into the Turbo Blaster.

Spazilton, 2020-12-17:
Have you attempted to put the Noir in Controller Passthrough mode?

treismac, 2020-12-18:
Yes.  Regrettably it did nothing.

prgreadonly, 2023-05-02:
Having the same issue with my BDL Turbo Blaster and my Analogue NT Mini Noir (both stock firmware and jailbreak firmware).
Controller passthrough mode doesn't make a difference.
I also tried switching the "Four player settings" between standard/4score/fami4score but nothing seemed to make a difference
Glad to see it's not just me

tjanas, 2023-05-02:
Have you reached out to Analogue to see if they are aware of the issue, and when they except to release the next firmware update?

prgreadonly, 2023-05-02:
> Have you reached out to Analogue to see if they are aware of the issue, and when they except to release the next firmware update?

I just went ahead and sent them an email about this.

FWIW - I tested the BDL NES Turbo Blaster on my Analogue NT and it looks like the Analogue NT also fails to recognize input from the BDL turbo blaster

prgreadonly, 2023-05-03:
Just got a typical non-response from Analogue:

    Thanks for reaching out. Please note that Analogue consoles only support the official firmware.
 
    Thank you.
 
    Best Regards,
    Analogue Support
    [www.analogue.co](http://www.analogue.co/)

For what its worth, I had already mentioned that this occurs on the official firmware and the jailbreak firmware.
Really makes you feel like a moron spending $1000 on hardware from Analogue only to see them not even remotely support it.
MiSTeR is really the way to go from here on out - at least I would be able to fix this issue myself...

prgreadonly, 2023-05-03:
It may be worthwhile for someone else to try to contact Analogue - I'm not sure why they dismissed my email immediately.  I did reply and let them know (again) that the issue is present on official FW, but I'm not expecting to hear anything intelligent in repsonse

tjanas, 2023-05-04:
@prgreadonly @treismac A new official firmware was released today, can you please try this ASAP? They have said they fixed ["All known bugs fixed in Super Nt, Mega Sg, Nt mini Noir](https://twitter.com/analogue/status/1654144815226064896). https://www.analogue.co/support/super-nt/firmware/5.1

prgreadonly, 2023-05-04:
Thanks for letting me know!  I just tested my BDL Turbo Blaster on the new 1.3 NT Mini noir firmware, but the issue is still present.  The Analogue NT mini noir doesn't recognize any input from the BDL Turbo blaster.

It might be worth noting that the BDL turbo blaster does seem to be getting power: the turbo LEDs do blink, but no input is recognized by the NT mini (on all firmwares)

tjanas, 2023-05-04:
Does it work on an original NES?

prgreadonly, 2023-05-04:
> Does it work on an original NES?

Yup - works my original NES just fine.  I don't have a toploader to test at the moment, but absolutely no issues with the original NES

tjanas, 2023-06-26:
@treismac @prgreadonly Please re-test with JB6.7 and if it is still an issue, open a ticket at https://github.com/analoguejb/Analogue-Nt-Mini-Noir-JB

closed 2023-06-26 by tjanas

prgreadonly, 2023-06-27:
Thanks tjanas - went ahead and opened https://github.com/analoguejb/Analogue-Nt-Mini-Noir-JB/issues/4