SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #58: Vs. Tetris: unable to insert coin

tjanas, 2020-12-17:
The DIP Switch Core menu option is not available when loading this rom that is padded for Nt Mini compatibility. I believe it is using correct and complete NES2.0 headers.
[image]

acidrain97, 2020-12-19:
Vs. Tetris has only 24KiB of PRG-ROM but the jailbreak will only load a ROM with 32KiB of PRG-ROM. You can add 8,192 padding bytes of 0x00 bytes (or 0xFF bytes) inserted between the header and start of the PRG-ROM to get the game to work. Vs. Gumshoe may also need padding in addition to ROM reordering (see NewRisingSun's fixes).

This was in the read me file. It may help with Vs. Tetris.

tjanas, 2020-12-20:
I believe this was user error on my part.

closed 2020-12-20 by tjanas