SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #59: Intellivision games probably wrongly configured will freeze all input

Elrinth, 2020-12-18:
There Intellivision games will freeze all input on the analogue nt mini noir and just blink on the menu screen forcing you to restart your console. (i tried following the recommended numbers, where they were available on the  https://github.com/dot-bob/int2intv?fbclid=IwAR3_Wh1quSkuQcK8RqBF2DKBkty28UeUlc-NO1qUllau5qeKQQJq8grAGdM
:
all-star major league baseball
b-17 bomber
body slam - super pro wrestling (1988)
bomb squad (1982) mattel
bump 'n' jump
championship tennis
chip shot - super pro golf
commando
deep pockets - super pro pool and billiards
dig dug
diner
doom - not starting
game factory prototype
hover force
intellivoice bios
jetsons
king of the mountain
magic carousel
masters of the universe - the power of he-man
melody blaster
mind strike!
mr. basic meets bits n bytes (all aversioner)
Number Jumble
pole position
slam dunk - super pro basketball
Song player - ALL
Spiker - Super Pro Volleyball
Spirit V1.0 (2003)
Stadium Mud Buggies (1988)
Super Pro Decathlon (1988) (Intv Corp)
Super Pro Football (1986) (Intv Corp)
Super Soccer (1983) (Mattel)
TRON - Solar Sailer (1982) (Mattel)
Thunder Castle (1982) (Mattel)
Tower of Doom (1986) (Intv Corp)
Tripple Challenge (1986) (Intv Corp)
Vectron (1982) (Mattel)
World Cup Football (1985) (Nice Ideas)

games I could not get working:
Doom
Utopia (pink color?)
pinball (black screen)
shark shark [!]
atlantis (simply grey screen**
grom - only pink screen (simply not work)
groovy only plays sample groovy all the time
go for the gold - only pink screen
dreadnaught factory, the [a1]
duncan's thin ice
league of light a2
mind hunter a1
minotaur v1.1
World Series Major League Baseball (1983) (Mattel)

tjanas, 2020-12-18:
I can recall playing Dig Dug and Bump & Jump on the INTV Core without issue; are you using a Super Famicom NTT Data Controller so you can use the number pad functionality?

Elrinth, 2020-12-18:
I'm just using the SNES 8bitdo with retro reciever to see which games boot :)
Seems I maybe did something wrong with the conversion of those games then. I'll give a try again.

tjanas, 2020-12-18:
Check your converted files vs. the ones here and see if the checksums match. https://github.com/SmokeMonsterPacks/EverDrive-Packs-Lists-Database/blob/master/EverDrive%20Pack%20SMDBs/Analogue%20Nt%20Mini%20%26%20Noir%20Jailbreak%20SMDB.txt

tjanas, 2020-12-18:
I definitely know Diner works, I played it last night. 
[image]

Firebrandx, 2020-12-18:
The issue is when you don't have the proper BIOS file in the BIOS folder. For example, if you try to load an Intellivoice ROM without the 012.bin BIOS file, the game will not load, and you will not be able to load any other game after that. This is how the glitch works.

Elrinth, 2020-12-18:
Aha, so that is perhaps the problem.

Firebrandx, 2020-12-18:
Hmm well that may be a different glitch then. The one I had still allowed me to navigate the Noir menu, but I wasn't able to load any other game until I exited and re-entered the core.

Elrinth, 2020-12-18:
Ok #MAP1, #MAP2 and#MAP5 are wrongly coded in int2intv program. (I will make issue for the repo) if I am to believe how his code works...
see here @ for example line 99, it should be 0x3000 not 0x2FFF:
https://github.com/dot-bob/int2intv/blob/main/int2intv.cpp

The maps can all be found here:
http://spatula-city.org/~im14u2c/intv/dl/intv_maps.zip

And if to believe how "large" each dataset should be, they should be 0x1000 0x2000 0x3000 or so... But he type in 0x2FFF. So I will try to help him know about this problem. Also I will see what CRC32 I get after updated program is used!

Elrinth, 2020-12-18:
I'm closing this as this is fault on my side using broken games. While it obviously is desirable that the NT Mini Noir doesn't hang, this might be a requirement for the system to actually hang when loading roms. I don't know.

But I'll close this and I will continue this discussion over with int2intv program creator at his github as that was really the problem.

closed 2020-12-18 by Elrinth