SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #60: MegaDuck games not working

Elrinth, 2020-12-18:
boots but not working:
snake roy (will reboot when press start, after inputting password in password screen will reboot)
zip ball (crash when start or password screen)

show logo or nothing:
       LastWriteTime         Length Name                                   
       -------------         ------ ----                                   
2013-08-11     23:32         131072 2nd Space.bin                          
2007-10-16     23:40          65536 A Force.bin                            
2014-02-26     02:35         131072 Ant Soldier.bin                        
2007-10-16     23:40          65536 BF Tale.bin                            
2007-10-16     23:40         131072 Bstf Gtr.bin                           
2014-02-25     16:01         131072 Captain KK.bin                         
2014-02-27     04:31         131072 Duck Adventure.bin                     
2010-05-17     20:13          65536 Railway.bin                            
2007-10-16     23:40          65536 Snake Roy.bin                          
2014-02-26     02:59          65536 Worm Vstr.bin                          
2014-02-26     03:06          65536 Zip Ball.bin                           
2010-05-17     20:13          32768 Arctic Zone.bin                        
2010-05-17     20:13          32768 Brick Wall, The.bin                    
2010-05-17     20:13          65536 Commin 5 in 1.bin                      
2010-05-17     20:13          65536 Mega Duck 4 in 1 Game.bin              
2010-05-17     20:13          32768 Street Rider.bin                       
1996-12-25     00:32          65536 Suleimans Treasure (1993)(Commin).bin

GreatHierophant, 2020-12-25:
Not a bug, should be closed.

Elrinth, 2020-12-25:
Just like GreatHierophant says, the files need to have proper filenaming for them to work. I've tried this and it works perfectly now. Thank you!

closed 2020-12-25 by Elrinth