SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #62: Headphone jack and NES core audio issues

The-Iron-Goat, 2020-12-18:
Headphone jack: L/R channels are swapped. They're fine with the HDMI and phono outputs.

Square wave bug: Very obvious in the stage 3/4 music of Trojan. Attached is an audio capture demonstrating the problem. The first part is from a modded NES, the second is from the Noir with the latest firmware.

The-Iron-Goat, 2020-12-18:
Trouble attaching the file, here's a link:

https://www.mediafire.com/file/3lmz98q4egbx6oh/Trojan.flac/file

tjanas, 2023-06-26:
@The-Iron-Goat Please re-test with JB6.7 and if it is still an issue, open a ticket at https://github.com/analoguejb/Analogue-Nt-Mini-Noir-JB. Per JB6.7 changelog:
- Added headphone Jack L/R Swap to Audio - Headphones (note will swap RCA jacks to opposite of their labeling)

closed 2023-06-26 by tjanas