SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #64: Top line missing in Game Gear graphics:

Firebrandx, 2020-12-19:
So this one had tricked me into thinking the scaler was bugged, but it turns out to be a different kind of bug:

The top line of graphics is blacked out, but still counts against the scaler. I confirmed it with an emulator screen grab.

acidrain97, 2020-12-19:
Speaking of the GG core is this the same one used in the SG Mega? If so it would explain a few things.

tjanas, 2021-02-23:
@Firebrandx can you confirm if this is fixed with the latest JB v6.6?

acidrain97, 2021-03-10:
what PCU do you recommend? Some like a 

Triad WSU120-1000-13?

Firebrandx, 2021-03-10:
Yeah that's the one I've ordered. Exact same specs as the shipped PSU.

acidrain97, 2021-03-16:
> Yeah that's the one I've ordered. Exact same specs as the shipped PSU.

Got the Triad.. Don't see a difference but if it prevents problems down the road works for me. THX for the OCD settings videos. What a difference things make with the proper settings.

Firebrandx, 2021-07-02:
This bug is confirmed fixed. Only minor non-issue is 720p vertical default centering is off (likely overlooked when coding the new firmware), but this is easily corrected by adjusting the value and saving the new settings.

closed 2021-07-02 by Firebrandx