SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #65: Idea/proposal: create new repo for Nt-Mini-Legacy-Jailbreak

tjanas, 2020-12-19:
The existing latest JB2.0 and prior versions exist [here](http://blog.kevtris.org/blogfiles/). For completeness and preservation, copies of these firmwares could be kept in a new repo. Outstanding issues with the original Nt Mini JB firmware could be documented too. I would like to document them before I forget, and perhaps Kevtris could address them when he gets some free time down the road.

frederic-mahe, 2020-12-20:
@tjanas if you'd create such repository under your user name, we could refer to it in the Noir README.

tjanas, 2020-12-22:
@frederic-mahe I have created it [here](https://github.com/tjanas/Nt-Mini-Legacy-Jailbreak).

closed 2020-12-22 by tjanas

frederic-mahe, 2020-12-22:
Thanks @tjanas I've added a link (see f7ca7ed8bb7a41ea58e78c428f6259cc8681182b)