SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #66: Music issue in Trojan (nes)

StephanKuwabara, 2020-12-20:
I've noticed a bit of an audio issue on the second stage of Trojan on the jail break.
It sounds like one or both of the pulse waves keeps fading out, which doesn't happen when I play on the everdrive on my NES.
I've tested it on the jailbreak, I've tested it with the everdrive on the Noir, and I've tested it with the everdrive on the NES.  In both cases on the Noir/Jailbreak the audio issue occurs, on the NES it plays normally.  I'm not sure if I should be bringing this up here or to Analogue.

Elrinth, 2020-12-30:
This is a duplicate of: https://github.com/SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak/issues/62

I have recorded footage of this bug here:
https://www.youtube.com/watch?v=BfrxJT6U90o

tjanas, 2021-02-23:
Please retest with JB v6.6

StephanKuwabara, 2021-02-23:
I did a complete playthrough last night on the latest firmware, and can confirm the fix works

closed 2021-02-23 by StephanKuwabara