SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #69: Issues with GBC games

sincethenes, 2020-12-20:
Two I’ve found so far. In Jungle Book: Mowgli’s Wild Adventure, if you pause the game, once you unpause the only thing visible onscreen is the Mowgli’s sprite. The rest of the screen is just a light brown color. You can fix in game by pausing again and backing out to the world map to restart the level, but then all progress is erased. 

Another issue is in Klax. Once you reach level five, every time you complete half the level, it crashes and shows the screen attached. 
[image]

matthargett, 2020-12-21:
can you file the issues separately (eg move the Klax one to a new issue, and rename this one)? also, can you post the md5 hash of the ROMs you used? thanks!

tjanas, 2021-02-23:
Please retest with JB v6.6

Elrinth, 2021-02-23:
For me it works atleast in latest version.

tjanas, 2021-02-24:
@sincethenes please confirm it is fixed so this issue can be closed

sincethenes, 2021-02-24:
No, it still fails for me. Attached are some screenshots. Jungle book after pause/Klax now during level 2
[image]

sincethenes, 2021-02-24:
[image]

sincethenes, 2021-02-24:
[image]

tjanas, 2021-02-24:
Can you confirm that you updated your core files when you updated the firmware?

sincethenes, 2021-02-24:
I’m on a set right now. How do I check, and I’ll do it later tonight?

tjanas, 2021-02-24:
@sincethenes make sure you download the latest updated [SD System Card Structure](https://github.com/SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak/blob/main/firmware/SD%20Card%20System%20Structure.zip) and extract those so they overwrite the existing ones on your sd card.

sincethenes, 2021-02-25:
I’m getting a 404 error on that link @tjanas . Also from the download link to the SD System Card Structure on the National-Mini-Noir-Jailbreak page.

sincethenes, 2021-02-25:
Not National, NT. (Autocorrect

tjanas, 2021-02-25:
> I’m getting a 404 error on that link @tjanas . Also from the download link to the SD System Card Structure on the Nt-Mini-Noir-Jailbreak page.

Try again.

sincethenes, 2021-02-25:
That did it. All fixed!

closed 2021-02-25 by tjanas