SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #71: RetroUSB 8-BIT XMAS 2016 glitched

tjanas, 2020-12-21:
Using https://retrousb.com/downloads/xmas16demo.nes after running NES2.0 XML script to modify the header to a correct header. Graphics on the Nt Mini Noir are glitched with this rom.
[image]

GreatHierophant, 2020-12-24:
They're glitched period, that is a bad dump.

tjanas, 2020-12-25:
Weird that that creator of the game would host a bad rom dump.

closed 2020-12-25 by tjanas

spspspspspsp, 2021-01-27:
I know this issue was closed, but I am also curious about this.

Is this really a bad dump?

It looks like could be incomplete, based on retroUSB's YouTube videos, but [this page](https://www.retrousb.com/product_info.php?cPath=31&products_id=152&osCsid=d8851490489c375d51fac5d462941fb2) also says this ROM is "for archival purposes only. No emulator support."