SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #74: NES:Famicom game~B-Wing ,Saint Seyia-Ougon Densetsu ,Dynamite Batman

box1694, 2020-12-22:
on SD card:
B-Wing : get blue Screen
Saint Seyia-Ougon Densetsu : graphics broken(it's same on n8pro)

flashcart:
Dynamite Batman : can't use flashcart n8 n8pro

Thanks your work!

tjanas, 2020-12-22:
Can you inspect these roms with [NESExplorer](https://www.romhacking.net/utilities/1425/)?
What is the CRC32 checksum for your **B-Wings (Japan).nes** file?
It should be `d0f54d24` when using this correct header:
`4E 45 53 1A 02 01 90 B8 70 00 00 00 00 00 00 01`
Other info:
```
B-Wings (Japan).nes
===================
PRG: 2 (32KB)
CHR: 1 (16KB)
Mapper: 185
Mirroring: H
Format: NTSC
* 4-screen mirroring

Detailed information:
---------------------
MD5          : 75bed5e623febf824bb3b26e1310cb63
CRC32        : d0f54d24
PRG CRC32    : 725d5cc4
CHR CRC32    : 456df683
PRG+CHR CRC32: 74f0a89f
```

**Saint Seiya - Ougon Densetsu (Japan).nes**:
Header: `4E 45 53 1A 08 10 80 98 00 00 00 00 00 00 00 01`
```
Saint Seiya - Ougon Densetsu (Japan).nes
========================================
PRG: 8 (128KB)
CHR: 16 (256KB)
Mapper: 152
Mirroring: H
Format: NTSC
* 4-screen mirroring

Detailed information:
---------------------
MD5          : d2b1d68bc7814547dfc64d219cd938e2
CRC32        : 74641d58
PRG CRC32    : 3bf15767
CHR CRC32    : e7c63c56
PRG+CHR CRC32: 026c5fca
```

From the XML database mentioned in the JB readme:
```
	<game><!-- Licensed Japan\B-Wings.nes -->
		<prgrom size="32768" crc32="725D5CC4" sha1="D3F89EE488C07838D4D99E92C0EF38B5335FC174" sum16="6673" />
		<chrrom size="8192" crc32="456DF683" sha1="DEFA1F639D3248D6C366986D17F3F8634D25F036" sum16="DBE5" />
		<rom size="40960" crc32="74F0A89F" sha1="1862126A3D8B8E37AC0571AA76309D0BB2DD8DB2" />
		<pcb mapper="185" submapper="7" mirroring="H" battery="0" />
		<console type="0" region="0" />
		<expansion type="1" />
	</game>
	<game><!-- Licensed Japan\聖闘士星矢꞉ 黄金伝説.nes -->
		<prgrom size="131072" crc32="3BF15767" sha1="1CA2EE893FE00C942822E00DBDC3D0093380A5BA" sum16="CDDD" />
		<chrrom size="131072" crc32="E7C63C56" sha1="788A455EC75B54686B0CA50A3AADEFC43CBE7B4F" sum16="9B55" />
		<rom size="262144" crc32="026C5FCA" sha1="EEB560ADE82C5C322CD515176E9E9B6F4847B0A6" />
		<pcb mapper="152" submapper="0" mirroring="H" battery="0" />
		<console type="0" region="0" />
		<expansion type="1" />
	</game>
```

tjanas, 2020-12-22:
Note that the two listed above required rom hacks and/or mapper changes to be able to run correctly on old flash carts like the EverDrive.

tjanas, 2020-12-23:
@box1694 see above

box1694, 2021-01-02:
B-Wings (Japan).nes
===================
PRG: 2 (32KB)
CHR: 1 (16KB)
Mapper: 185
Mirroring: H
Format: NTSC
* 4-screen mirroring

Detailed information:
---------------------
MD5          : 1b24f09725092a2526fde24426eb6bc8
CRC32        : 15a4facb
PRG CRC32    : 725d5cc4
CHR CRC32    : 456df683

Saint Seiya - Ougon Densetsu (Japan).nes
========================================
PRG: 8 (128KB)
CHR: 16 (256KB)
Mapper: 70
Mirroring: H
Format: NTSC
* 4-screen mirroring

Detailed information:
---------------------
MD5          : 786c3e5c86ece7e4f34ac10646b5fe37
CRC32        : d122ba8d
PRG CRC32    : 3bf15767
CHR CRC32    : e7c63c56
PRG+CHR CRC32: 026c5fca

tjanas, 2021-01-02:
Looks like your rom data is good but your NES headers are bad. Run the NES2.0 header script per the instructions in the readme

closed 2021-02-25 by tjanas