SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #76: Kick Master - Wavy effect in Title Screen if 16 Sprites Per Scanline is enabled

MP2E, 2020-12-23:
Here's a video of the effect, I also show that it looks correct if 16 Sprites Per Scanline is disabled

https://streamable.com/mylyik

MP2E, 2020-12-23:
I think this is a bug, as this does not occur on MiSTer with its 16 sprites per scanline option. Not sure about other consoles/emulators

BobsYurUncle, 2021-01-08:
I found this same bug on the HUD during the sidescroller sections of Vice: Project Doom

https://user-images.githubusercontent.com/37429660/103980563-f37ae980-5134-11eb-8c94-27bd8c7e5aad.mp4

BobsYurUncle, 2021-01-11:
Same with Magician on the text bar.

https://user-images.githubusercontent.com/37429660/104143557-d92f4e80-5374-11eb-8972-ca3b92d850e3.mp4

MP2E, 2021-02-22:
fixed by NT Mini Noir v1.2 firmware, I assume this will be in a jailbreak soon

closed 2021-02-22 by MP2E