SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #77: Battletoads Double Dragon P2 Character Select glitch

sponge9012, 2020-12-23:
When you reach the player select screen, player 2 will start moving randomly as if with ghost inputs. After selecting characters and starting the game, everything seems normal.

On jailbreak 6.5 firmware, using a 2.0 headered rom. i dont have a cart to try. tested with 8bitdo 2.4 and bluetooth controllers, as well as a regular official wired controller.

tjanas, 2020-12-23:
@sponge9012 please provide your NES2.0 header as well as checksum information.
Can you confirm the following matches for your `Battletoads-Double Dragon (USA).nes` file:

CRC32: 26de4e7b
NES2.0 header: `4E 45 53 1A 10 00 70 08 20 00 00 07 00 00 00 01`

sponge9012, 2020-12-23:
i can confirm the checksum matches using nesexplorer. how can i check the header string easiest

tjanas, 2020-12-23:
If the checksum matches, then your header is good.

GreatHierophant, 2020-12-24:
Passthrough Mode keeps the 2nd player selector from randomly moving.

tjanas, 2021-02-23:
Can you retest with JB v6.6?

sponge9012, 2021-02-23:
yes, i just did and the glitch is gone. thank you!

closed 2021-02-23 by sponge9012