SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #80: Famicom Network Controller HVC-051 implementation for CV and INTV cores?

chopsticksamurai, 2020-12-27:
I recently picked up one of these controllers to use with the Colecovision and Intellivision cores.  I see that this part of the Readme for this controller is now struck out.  Will this controller be implemented in the future?  I specifically bought one of these since it was originally listed as being compatible.  Thanks in advance!

GreatHierophant, 2020-12-28:
Make sure you overwrite all the *.nt2 files in the \SYSTEM\CORES directory on the SD card with the ones from the latest jailbreak.

chopsticksamurai, 2020-12-28:
> 
> 
> Make sure you overwrite all the *.nt2 files in the \SYSTEM\CORES directory on the SD card with the ones from the latest jailbreak.

All I am seeing is the .bin file for installing the jailbreak.  How would I go about overwriting the files?  Thanks!

tjanas, 2020-12-28:
@chopsticksamurai https://github.com/SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak/blob/main/firmware/SD%20Card%20System%20Structure.zip

tjanas, 2020-12-28:
Per the README:
Step 1 is the .bin file
Step 2 is the SD System Card Structure

> **Step 2: System Setup**
> Unzip SD Card System Structure.zip into the root directory of your SD card. You will now have a set of folders reflecting where game ROMs should be stored. Follow the instructions below for configuring the /BIOS/ folder.

chopsticksamurai, 2020-12-28:
@tjanas @GreatHierophant Thanks so much for pointing this out.  I was under the impression that the structure zip was just a collection of empty folders to help you set up the SD card.  It's not clear that there are updated system files in there as well.  It's also confusing having the info about this controller struck out in the README as well.  But all is working as it should now. 👍 Thanks for your help!

closed 2020-12-28 by chopsticksamurai