SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #81: SNES NTT Data Controller Issue

Sparkoids, 2020-12-28:
Everything except the hash # key works. 

I have tested the controller in an Analogue Super NT using a tester and it's perfect but in the Colecovision implementation the key doesn't register and all the others do.

GreatHierophant, 2020-12-28:
I do not have an issue with my NTT Data Pad.  I tried Boulder Dash, which requires pressing the # key to start the game or restart a level and it worked fine on both occasions.

Sparkoids, 2020-12-29:
I will try that - it doesn't display on the Joystick tester - maybe it's that program that is the issue...

tjanas, 2021-01-01:
I just tried Boulder Dash on Coleco and the # button on the Super Famicom NTT Data Controller (NDK10) works fine.

eewokee, 2021-01-30:
Works fine for me too.

tjanas, 2023-05-04:
Can't be reproduced by @GreatHierophant, @eewokee, or @tjanas 
Possibility a bad controller @Sparkoids ?