SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #85: CopyNES Generic MMC3 - GUN-NAC dump

onthewaveagain, 2021-01-01:
I tried to dump my original cartridge of Japanese GUN-NAC with the Generic MMC3 module. There is no battery in this game.

It worked and the ROM seems to play well. I have however noticed a weird issue during the intro. See here [image] I have tested the ROM with my NT mini Noir and with RetroArch (FCEUMMcore) and get this on both.
When played with my original cartridge, the text is all white.

Dumped file hash : E5354A5FB794DD1DBE92685F55B5342ACE09CAC6BEFB2444ECE557FF4D19CCF7

It seems to be very minor but I thought I would report.

tjanas, 2021-01-01:
There is a bug with CopyNES on the current Noir JB firmware, where it will corrupt the dump of the PRG ROM portion of the nes dump.

onthewaveagain, 2021-01-03:
It is sadly a regression then, Hopefully this will be fixed. I just tested with my jailbroken NT Mini and I don't get that glitch.

onthewaveagain, 2021-03-27:
Fixed with JB6.6

closed 2021-03-27 by onthewaveagain