SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #87: Weak microphone sensitivity

tjanas, 2021-01-03:
I’ve found that even with the microphone setting set to the maximum, the microphone is still not verify sensitive; in the Japanese Zelda 1, it still requires strong blowing into the microphone to kill Pols Voice. On the original Nt Mini, setting the microphone sensitivity at or near maximum would always pick up enough noise just from the background, offering a fuller range of microphone sensitivity values.

Elrinth, 2021-01-03:
I felt that on the original Famicom, killing "Pol's voice" required multiple vocal tries.

tjanas, 2021-01-04:
I tried two microphones... one is an old SoundBlaster PC mic, the other being the in-line headphone/mic that came with my PS4