SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #89: Battletoads and Double Dragon Abobo Fight Crash

BobsYurUncle, 2021-01-08:
This one is easy to reproduce. Using 6.5 Jailbreak firmware and the rom from no-intro romset on Archive.org. Sandisk U3 MicroSD.

Play until level 1-2, after you defeat the two ball-hand enemies, Abobo crashes through a metal wall. At this point the game freezes for me, sometimes with full screen graphical glitches.

I tried about 6 times playing as Pimple and Billy across 1080p and 480p. It froze every time except once when I was playing as Billy at 480p resolution, not sure why but I was able to proceed the game.

Rom Info:

Detailed information:
---------------------
MD5          : 35933222cf8658f7c6679fc7de630aaa
CRC32        : 53b02f3d
PRG CRC32    : ceb65b06
PRG+CHR CRC32: ceb65b06

6C 13 00 4C 2F EC 4C 19 83 4C 2F EC 4C 26 8E 4C

I will also post this thread I found of a fella experiencing the same issue, playing off his Power Pak:

https://forums.nesdev.com/viewtopic.php?t=15222

[image]

tjanas, 2021-01-08:
@BobsYurUncle you must have a bad rom if your NES header is `6C 13 00 4C 2F EC 4C 19 83 4C 2F EC 4C 26 8E 4C`
In fact, I'd bet money that you have an unheadered rom, and the first PRG bytes after the header match those bytes above. That also means you didn't even run the NES Header Script, because it would have corrected an otherwise good unheadered rom for you.

tjanas, 2021-01-08:
@BobsYurUncle your [header](https://wiki.nesdev.com/w/index.php/NES_2.0) bytes for this game should be the following (on a side note, all headered NES roms should always begin with bytes `4E 45 53 1A`, which is "NES<EOF>"):
`4E 45 53 1A 10 00 70 08 20 00 00 07 00 00 00 01`
```
	<game><!-- Licensed North America\Battletoads &amp; Double Dragon꞉ The Ultimate Team.nes -->
		<prgrom size="262144" crc32="CEB65B06" sha1="A14563325B0F33C358142E7363D31614722FDDB1" sum16="3135" />
		<rom size="262144" crc32="CEB65B06" sha1="A14563325B0F33C358142E7363D31614722FDDB1" />
		<chrram size="8192" />
		<pcb mapper="7" submapper="2" mirroring="H" battery="0" />
		<console type="0" region="0" />
		<expansion type="1" />
	</game>
```

BobsYurUncle, 2021-01-09:
So it's not so much a "bad rom", rather an unaltered one. You'll have to forgive me not reading the encyclopedia of instructions, I typically run my games off cartridge. I had no idea this header requirement existed because in all the coverage of the nt mini I've watched, nobody has mentioned it. 

Seems like a pretty darn important thing and I would suggest updating the Readme to put a note about "Rom Setup" right under "System Setup" to help avoid false positives like mine.

I consider myself fairly technical and this went unbeknownst until now. Also that repair script seems like its got problems of its own because I got a "bad header" error the first time I ran it, second time fixed it though. So all good over here. Sorry for not reading but c'mon, that thing is a novel.

closed 2021-01-09 by BobsYurUncle