SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #92: (INTV Core) Memory appears to persist when a new ROM is loaded, causing errors

spspspspspsp, 2021-01-21:
I have been exploring the Intellivision Core, and in the process have been slowly converting homebrew titles I have collected to the .intv format.

While testing, I noticed that Stack 'Em! will sometimes not load correctly (and thus not play normally). This appears to depend on the last ROM that was loaded.

For example, if I load D1K Arcade to the title screen, then load Stack 'Em!, DK's tie is visible in Stack 'Em!.

Stack 'Em! is the only game I have experienced this issue with so far, and being a relatively old homebrew title, perhaps fault lies in the way this ROM was coded, but I don't normally expect the contents of memory to persist like that.

For reference, here is how I converted the memory banks using int2intv:

// Stack 'Em!
MapAddress[0] = 0x4800;
DataSize[0] = 0x0800;

// D1K Arcade
MapAddress[0] = 0x5000;
DataSize[0] = 0x2000;

MapAddress[1] = 0x9000;
DataSize[1] = 0x3000;

MapAddress[2] = 0xD000;
DataSize[2] = 0x1000;

MapAddress[3] = 0xF000;
DataSize[3] = 0x1000;

[image]
[image]

[Stack 'Em! (2004).zip](https://github.com/SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak/files/5846201/Stack.Em.2004.zip)

embusalacchi, 2021-01-21:
I don't know how to fix the problem but if you ever publish your work in converting the home-brew titles into .intv format I would love to try them out.

spspspspspsp, 2021-01-21:
Also, for those interested, there is some homebrew that is incompatible with the Noir. For example, Grail of the Gods requires JLP functionality (on the LTO Flash cart and built into some emulators), and will hang during dungeon generation.

tjanas, 2023-06-26:
@spspspspspsp Please re-test with JB6.7 and if it is still an issue, open a ticket at https://github.com/analoguejb/Analogue-Nt-Mini-Noir-JB. Per JB6.7 changelog:
- Fixed Intellivision RAM not cleared before loading a game

closed 2023-06-26 by tjanas