SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #93: No button is mapped to "0" in the INTV core on the Famicom Network Controller

spspspspspsp, 2021-01-27:
Since "0" was remapped to ".", no button is mapped to 0 in the INTV core on the Famicom Network Controller.

When playing Ultimate Pong, for example, you are unable to leave the options menu because there is no key for "0".

Can "0" please be mapped to the "End" button on this controller?

spspspspspsp, 2021-01-29:
I converted the Hand Controller Diagnostic ROM earlier and looked at the Famicom Network Controller. I confirmed that no button is mapped to "0".

Also, the "." button is mapped to Left + Left fire + 0 + Enter:

[image]

tjanas, 2023-06-26:
@spspspspspsp Please re-test with JB6.7 and if it is still an issue, open a ticket at https://github.com/analoguejb/Analogue-Nt-Mini-Noir-JB

Per JB6.7 changelog:
Improved Intellivision's third fire button is now mapped to Famicom Network Controller's 0 button for easier access, Intellivision numberpad 0 button has been remapped to Famicom Network Controller's . button

closed 2023-06-26 by tjanas