SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #94: (INTV Core) Does the INTV2 format support RAM?

spspspspspsp, 2021-01-28:
There are some games that have a defined RAM address range.

Does the INTV2 format support the definition of RAM ranges?

In some cases, when I ignore this, it doesn't seem to impact the game, but I was testing Missile Domination today, and the game crashes whenever you start the game or enter the options menu. I was double-checking my address definitions (looked OK) and started wondering about RAM.

Missile Domination is mapped like so:
[mapping]
$0000 - $1DFF = $5000
$1E00 - $29FF = $D000
$2A00 - $39FF = $F000
[memattr]
$8000 - $81FF = RAM 16

And I converted the game to .intv using the following definition in int2intv:
MapAddress[0] = 0x5000;
DataSize[0] = 0x1E00;
MapAddress[1] = 0xD000;
DataSize[1] = 0x0C00;	
MapAddress[2] = 0xF000;
DataSize[2] = 0x1000;

Or, does the RAM address simply not matter?

spspspspspsp, 2021-01-28:
Wait, re-reading the INTV2 description I see a bit about RAM.

Adding the following may work.

_MapAddress[1] = 0x8000;
DataSize[1] = 0x0200;_

Then the docs say bit 16 is used to indicate whether a range is writable. Does that mean bit 16 of "0x8000" or bit 16 of "00 80 00 00" in the binary?

spspspspspsp, 2021-01-28:
I got a little further by adding a bank like /00 80 00 00\ /00 02 00 00\ <zeroed data> at the end of the file and after the ROM bank at $5000.

I tried flipping bits from 0 to 1 in this address/length declaration, and when, for example, I use  /10 80 00 00\ /00 02 00 00\ I can access the options menu, but I cannot set options, nor can I see the cursor or missiles during gameplay.

How does the RAM declaration work?

tjanas, 2023-06-26:
@spspspspspsp I do not know the answer to this question, but you may have luck at the repo where the latest JB firmware is being hosted: https://github.com/analoguejb/Analogue-Nt-Mini-Noir-JB

closed 2023-06-26 by tjanas