SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #95: Zelda 1 FDS broken on OFW 1.2

Peredonov, 2021-02-22:
Tested on RAM Adapter + FDSStick that tests and performs OK on real Famicom.

After loading game and copyright screen, title screen appears graphically garbled in varying ways, such as this one:

https://media.discordapp.net/attachments/446407171471835136/813518219573723136/IMG_7682.JPG

Changing cartridge IO to 3.6 leads to a black screen after the copyright screen.

ZReport, 2021-02-25:
Whatever is breaking this also screws up Genesis core games, such as Sonic & Knuckles, Sonic 3, Streets of Rage 2 & and more.

tjanas, 2021-02-25:
> Whatever is breaking this also screws up Genesis core games in the JB6.6, such as Sonic & Knuckles, Sonic 3, Streets of Rage 2 & and more. It also throws glitches and errors in a few SMS games too.

Can you create a separate ticket for that issue?

tjanas, 2021-02-26:
Per @GreatHierophant on Classic Gaming discord:
> FDS games appear to be glitchy on the new firmware, whether you run them on an FDS, an N8 or an N8 Pro (if you can).

Peredonov, 2021-03-07:
It appears that turning on dejitter allows this game to run properly, and it also lets the FDS RAM adapter pass the self test. We can assume kevtris had dejitter enabled during testing.

I suppose this issue should stay open however, as I do not know if dejitter can have its own disadvantages.

tjanas, 2021-03-07:
I agree it should stay open until it works correctly without dejitter enabled.

qcretro, 2021-03-10:
I can confirm this issue with 2 different known working HVC-FMR-04 RAM adapters. I am testing using Metroid (Rev 2 & 3). Works and loads correctly with OFW 1.1 or JBFW 6.5, not with OFW 1.2 or JBFW 6.6. Using default settings, except cartridge audio turned on. Dejitter disabled.

tjanas, 2023-05-04:
@Peredonov @qcretro please re-test your FDS RAM Adapters with the latest official firmware update released today https://www.analogue.co/support/nt-mini-noir/firmware/1.3

closed 2023-05-06 by tjanas