SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #97: CopyNES Mini Backing up ROM but not Save Files

neworion, 2021-02-24:
I successfully made playable backups of all of my NES carts. Each rom works, even the ones I mention below with battery saves. Yet the save files themselves appear to be corrupt.

Dragon Warrior and Zelda 1 & 2 each used the MMC1 chip. While CopyNES created 3 *.sav files that were 8kb in size, they are not recognised by the NT mini noir or by Mesen NES emulator on the PC. So yes I did select the battery option while saving 😉

I tried using Jailbreak versions 6.5 & 6.6 and neither worked. I would love to get these save files backed up.

tjanas, 2021-02-24:
Duplicate issue. See https://github.com/SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak/issues/11

whitezombie2000, 2022-02-13:
This issue should be closed.

closed 2022-02-14 by tjanas