SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #101: Crystalis / God Slayer can't move the player character in beginning of game

Elrinth, 2021-03-01:
"Has anyone noticed Crystalis (US) having issues after updating their Nt’s firmware?  It now hangs after the name entry screen.  Testing on an N8 Pro with latest firmware." - DigitalGamer @ discord
"Game is totally broken for me.  The character just shakes on the screen." - DigitalGamer @ discord

"just tested with nt mini jb nes core from sd card and same thing" - samplya @ discord
"After name entry screen I see the main character blinking in a doorway and cannot do anything" - sampyla @ discord

This has been confirmed problematic by me on God Slayer cartridge (japanese game).

Jeffersonian, 2021-03-01:
This is a regression in the 1.2 retail firmware; the issue does not occur in 1.1.  This also reproduced for me when resuming an old save state and transitioning to a new screen (the character freezes like after the start of the game).

Jeffersonian, 2021-03-01:
Per discord discussion, this only occurs if two or more controllers are plugged in.  This does not occur if passthrough mode is enabled.

whitezombie2000, 2022-02-16:
I was able to reproduce this issue with a real cartridge of Crystalis having two controllers inserted. This issue occurred on the official 1.2 firmware and 6.6 JB firmware. I was able dump the ROM with Generic MMC3 TxROM and the same issue occurred running the ROM with two controllers inserted. The issue does not occur if Passthrough Mode is enabled an mentioned above. It should be noted the back of my cart says REV A.

Has this issue been submitted to Analogue Support?

tjanas, 2022-02-16:
@Elrinth and @Jeffersonian have you reported this issue to Analogue? If not, please report it.

whitezombie2000, 2022-02-16:
I sent Analogue Support an email just to make sure it got reported with a description of the problem, and a link to this GitHub post.

Jeffersonian, 2023-05-04:
I have confirmed that this is fixed in the 1.3 retail firmware.

tjanas, 2023-05-04:
Thank you for testing @Jeffersonian. I am going to close this issue. In the unlikely event the bug exists in a future JB, it can be re-opened.

closed 2023-05-04 by tjanas

whitezombie2000, 2023-05-04:
Wow, they release a new firmware. I didn't think they would. I guess I'll have to check it out. I was getting ready to sell my Noir.

whitezombie2000, 2023-05-05:
I can confirm this as well with Crystalis. No issue on v1.3 firmware with two controllers plugged in.