SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #103: Original Everdrive N8 doesn't work with JB Firmware v. 6.6

jengelz, 2021-03-05:
Beside my case, Smokemonster informed me several cases are reported. My original Everdrive N8 doesn't work anymore, after updating from jailbreak firmware v. 6.5 to v. 6.6. I have vertical dotted lines in the Everdrive Menu. When booting games, there is just a completely grey screen, i tested a few roms. The Everdrive N8 Cartridge still works fine in my original NES System. So the Cart is fine. On the Reddit Support Megathread, i had a comment, the N8 Pro seems to work though.
[image]
[image]

jengelz, 2021-03-05:
The Everdrive is on Firmware v. 1.25, the latest update from Krikzz

[image]
[image]

tjanas, 2021-03-05:
Can you try downgrading the EDN8 firmware to version 1.16? In the past, I have had various issues with EDN8 firmware after that version on a real NES front loader.

jengelz, 2021-03-05:
Nope, even worse. I heard about going back to versions before 1.20. But in this case, no improvement. The N8 Cart with v.1.16, i've tested again beforehand on my NES Frontloader, loaded a rom, and it worked. So this screen then on the NT Mini Noir
[image]

qcretro, 2021-03-10:
I can confirm the same behavior with the original Famicom N8 on the latest FW.

tjanas, 2023-05-04:
@jengelz @qcretro please re-test your N8 cartridges with the latest official firmware update released today https://www.analogue.co/support/nt-mini-noir/firmware/1.3

jengelz, 2023-05-06:
Hi Terry,

A guy on Twitter reported to me, he could run the Original Everdrive N8 (so not the Pro) with yesterday’s firmware update.
I can forward you the tweet if you like.
So the bug seems to be fixed (though it is not mentioned in the changelog)

I am awaiting the Jailbreak from the newest firmware, i really like the cores,

If you still like me to do the test, i can do it for you, just let me know!

Cheers guys,

Jan

tjanas, 2023-05-06:
Thank you, @jengelz . I will report this issue as closed, but if you have a link to the tweet, it would be greatly appreciated if you could provide it here.

closed 2023-05-06 by tjanas

jengelz, 2023-05-07:
Hi Terry,

Here is the tweet thread:

https://twitter.com/jengelenm/status/1654154758586789889

Cheers

***@***.***
***@***.******@***.***